mjansen4857/pathplanner #555: Link for C++ API documentation not working

rtsledge, 2024-01-14:
Link for the C++ API documentation doesn't appear to be working

**To Reproduce**
Click any of the C++ API documentation Links
or try to go here
https://pathplanner.dev/api/cpp/

mjansen4857, 2024-01-14:
It's working fine. If you're on a school network it's likely being blocked.

rtsledge, 2024-01-14:
hmm.. not on a school network.  I am using Chrome browser...  When I click "PPLib C++ API" from the "Getting Started" page 

[image]

It takes me to "https://pathplanner.dev/api/cpp/"

but that looks like 

[image]

rtsledge, 2024-01-14:
oh wait never mind... I am sorry....

closed 2024-01-14 by rtsledge