mjansen4857/pathplanner #565: Flipped Choreo auto path doesn't flip angular velocity

Codehc, 2024-01-20:
Take any Choreo generated path that has some rotation and use it in a PathPlanner auto. If you run it on blue (not mirrored) it runs fine. If you run it on red (mirrored) none of the rotation actually happens unless you use rotational feedback. Rotational feedforward doesn't work.

Video:
https://github.com/mjansen4857/pathplanner/assets/57022309/ff2c5c82-bd6c-47f3-a5ce-5159a5efad68

I cloned pplib and traced the bug to this line here:
`mirrored.holonomicAngularVelocityRps.ifPresent(v -> mirrored.holonomicAngularVelocityRps = Optional.of(-v));`

This checks if the *`mirrored`* path has the `holonomicAngularVelocityRps` field which it doesn't because it was just instantiated above. If you change this to *`state`* instead of mirrored for the check if it's present it works. I tested this in my own codebase. 

This is what it looks like with the change:
https://github.com/mjansen4857/pathplanner/assets/57022309/766541e6-7fc1-41bc-8925-702a62f77ae0

Codehc, 2024-01-20:
#566

closed 2024-01-20 by Codehc

mjansen4857, 2024-01-20:
2024.1.4 is currently being pushed with this fixed