mjansen4857/pathplanner #588: Numbers without a decimal point in JSON files crash GUI

jonahsnider, 2024-02-02:
**Describe the bug**

If you change any of the numbers in a JSON (ex. `.pathplanner/settings.json`) to no longer have a fractional value, PathPlanner will crash when it tries parsing the file.

**To Reproduce**

1. Open `.pathplanner/settings.json`
2. Change `defaultMaxAccel` to `3`
3. Open PathPlanner
4. Observe that the app crashes

**Expected behavior**

The app should not crash. The JSON spec [says that numbers may have a fraction part](https://www.rfc-editor.org/rfc/rfc7159#section-6), but are not required to.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Versions: (please complete the following information):**
 - OS: N/A
 - GUI Version: 2024.1.2
 - PPLib Version: N/A
 - PPLib Language: N/A

**Additional context**

See https://github.com/dart-lang/sdk/issues/46883 for the others experiencing the same issue with Dart JSON parsing

mjansen4857, 2024-02-02:
Can fix this, but do not change the default constraints manually, as the changes will not be propagated to the paths using them until the next time settings are changed.

jonahsnider, 2024-02-02:
Thanks! For our team, we noticed this issue just from having Spotless configured with JSON formatting enabled. It'd remove the trailing `.0`s and cause PathPlanner to crash. We ended up just adding the config JSON as an exception in our project.

closed 2024-02-05 by mjansen4857