mjansen4857/pathplanner #599: Generating trajectories from paths with 2 points fails

wrall, 2024-02-07:
**Describe the bug**
Generating a PathPlannerTrajectory from just two PathPoints ends up giving a trajectory that takes infinite time.

**To Reproduce**
Steps to reproduce the behavior:
Run:
```
    public static void generateTrajectory()
    {
        ArrayList<PathPoint> points = new ArrayList<PathPoint>();
        points.add(new PathPoint(new Translation2d(0.0, 0.0)));
        points.add(new PathPoint(new Translation2d(-1.0, 0.0)));

        generateTrajectory(points);
    }

    public static void generateTrajectory(ArrayList<PathPoint> points)
    {
        PathConstraints constraints = new PathConstraints(
            2.54,
            2.54,
            Math.PI / 2.0,
            Math.PI / 2.0);

        PathPlannerPath path = PathPlannerPath.fromPathPoints(
            points,
            constraints,
            new GoalEndState(0.0, Rotation2d.fromDegrees(0.0)));
        PathPlannerTrajectory trajectory = new PathPlannerTrajectory(path, new ChassisSpeeds(), new Rotation2d());

        double totalTimeSeconds = trajectory.getTotalTimeSeconds();
        if (Double.isInfinite(totalTimeSeconds) || Double.isNaN(totalTimeSeconds) || totalTimeSeconds < 0.0)
        {
            System.out.println("TotalTimeSeconds is not a valid number: " + totalTimeSeconds);
        }
        else
        {
            System.out.println("x,y,theta,dx,dy,omega");
            for (double time = 0.0; time <= totalTimeSeconds; time += 0.02)
            {
                State state = (State)trajectory.sample(time);
                double xPosition = state.positionMeters.getX();
                double yPosition = state.positionMeters.getY();
                double orientation = state.targetHolonomicRotation.getDegrees();
                double xVelocity = state.heading.getCos() * state.velocityMps;
                double yVelocity = state.heading.getSin() * state.velocityMps;
                double turnVelocity = state.holonomicAngularVelocityRps.isPresent() ? state.holonomicAngularVelocityRps.get() : 0.0;
                System.out.println("" + xPosition + "," + yPosition + "," + orientation + "," + xVelocity + "," + yVelocity + "," + turnVelocity);
            }
        }
    }
```
Outputs "TotalTimeSeconds is not a valid number: Infinity".  There is a workaround of adding an intermediary point, but this should not be necessary - the information above describes a very simple and valid trajectory.

**Expected behavior**
Output indicating the robot driving forward in the X direction for 1 meter, taking approximately 2 seconds.

closed 2024-02-07 by mjansen4857

irs1318ds, 2024-02-07:
OK, that makes sense.  I would suggest adding a comment like above to the Javadocs about the expectations for the PathPoint list ("a pre-smoothed path with enough samples to actually profile the motion").

wrall, 2024-02-07:
OK, that makes sense. I would suggest adding a comment like above to the Javadocs about the expectations for the PathPoint list ("a pre-smoothed path with enough samples to actually profile the motion").