mjansen4857/pathplanner #602: Path Planner command fails when launched from Shuffleboard.

rakar, 2024-02-08:
**Describe the bug**
Path Planner commands fail when run from Shuffleboard despite working when run from joystick.a().ontrue. Non-pathfinder commands that we try work either way, which is why I suspect that path planner lib (or our use of it) is involved. When the shuffleboard button is pressed, the command doesn't appear to run and the scheduler seems to fail until the robot code is reset.

**To Reproduce**
```
 private void configureBindings() {
    Set<Subsystem> set = new HashSet<Subsystem>();
    set.add(drivetrain);
    SmartDashboard.putData("Run Diagnostic",
        new DeferredCommand(() -> runDiagnosticTest(), set));
    joystick.x().onTrue(new DeferredCommand(() -> runDiagnosticTest(), set));
}

  public Command runDiagnosticTest() {
    Pose2d currentPose = MMField.getBluePose(drivetrain.getState().Pose);
    PathConstraints trajectoryConstraints = new PathConstraints(1.5, 3, 2 * Math.PI, 4 * Math.PI);
    List<Translation2d> bezierPoints = PathPlannerPath.bezierFromPoses(
        currentPose,
        MMField.getBlueWooferApproachPose(),
        MMField.getBlueWooferPose());
    PathPlannerPath path = new PathPlannerPath(bezierPoints,
        trajectoryConstraints,
        new GoalEndState(0, Rotation2d.fromDegrees(180)));
    path.preventFlipping = false;
    return AutoBuilder.followPath(path);
  }
```

**Expected behavior**
The DeferredCommand should regardless of how it is launched.

**Screenshots**

**Versions: (please complete the following information):**
 - OS: [e.g. Windows 11]
 - GUI Version: [e.g. 2024.1.1]
 - PPLib Version: 2024.1.6
 - PPLib Language: Java

**Additional context**

mjansen4857, 2024-02-08:
There is no difference from a command being scheduled from NT and being scheduled from a button, especially from PPLib's perspective. There is either something somewhere else in your code causing this or its a shuffleboard issue. Shuffleboard is not really maintained anymore and has a lot of issues. Try using Glass instead

rakar, 2024-02-08:
We also thought there was no difference other than, well, observed reality. We'll poke around some more.

closed 2024-09-13 by mjansen4857