mjansen4857/pathplanner #604: Paths lie outside field and robot moves erratically and spins out of control

DragonflyRobotics, 2024-02-09:
**Describe the bug**
Let me preface by saying that for specific reasons, we use a custom Kinematics calculator instead of SDK, but for all of Pathplanner 2023, we were able to get everything working flawlessly by simply converting the module positions to SwerveModulePositions that were fed into SwerveDriveOdometry. I will also clarify that our odometry works just fine. We are using a NavX for the gyro. 

***Issue***
We made a simple straight-line path upon enabling the robot, the path goes straight outside the field perimeter and the robot goes full speed in that direction. Other times, running the same path, the robot will spin out of control in random elliptical paths. Othertimes, it will merely stay stationary as the angle motors spin out of control. Right after auton, if we switch to telop, the robot runs perfectly without any issues. 

Our auton code replicates code from other teams and the example code provided as closely as possible (considering the kinematics changes). We are simply not sure where to even start debugging. Any help would be truly appreciated. 

Here is the repository for our current code: https://github.com/frc5417/FRC2024-5788
Here is the repository for our working 2023 code for reference: https://github.com/frc5417/CustomSwerve/

Our DriveBase.java contains most of the methods for supplying the configureHolonomic method while the actual AutoBuilder resides in AutonLoader.java

**Versions: (please complete the following information):**
 - OS: Windows 11 and Ubuntu 22.04
 - GUI Version: [e.g. 2024.1.4]
 - PPLib Version: [e.g. 2024.2.3]
 - PPLib Language: [e.g. Java]

**Additional context**
We have tried hard-setting the omega values to 0 to prevent rotation and tried many ways of resetting the starting odom position. As the robot is erratic and out-of-control during testing, we currently don't have any screenshots or videos, but please let me know if there is any particular information that would help resolve this.

mjansen4857, 2024-02-09:
I don't really have the bandwidth to debug other teams' projects, so you are likely better off making a Chief Delphi post or something. But, from a quick look through what really stands out to me is that the module order you are using to give positions to odom is the reverse of the module order that you are using when converting chassis speeds to module states, and actually setting those target states. All of the module orders need to be identical. See [here](https://github.com/frc5417/FRC2024-5788/blob/fde6dde5b4f0d353273467be424c63ec0e80cc9a/src/main/java/frc/robot/subsystems/DriveBase.java#L107), [here](https://github.com/frc5417/FRC2024-5788/blob/fde6dde5b4f0d353273467be424c63ec0e80cc9a/src/main/java/frc/robot/subsystems/DriveBase.java#L177), and [here](https://github.com/frc5417/FRC2024-5788/blob/fde6dde5b4f0d353273467be424c63ec0e80cc9a/src/main/java/frc/robot/subsystems/DriveBase.java#L184)

DragonflyRobotics, 2024-02-09:
That is an excellent point, I will try that out, and if not, I will post a question on Chief Delphi. Thank you so much!

mjansen4857, 2024-02-09:
Actually, it looks like you still have reversed module orders in a bunch of places in your code. You should have them all be in order from 0->3, not 3->0

DragonflyRobotics, 2024-02-09:
Well, my odometry works well from 3-->0 since the module order for our custom kinematics is from 0-->3. So all I did was change the module order when finding the SwerveModuleStates to 3-->0 to match our odometry. Is that incorrect?

mjansen4857, 2024-02-09:
> odometry works well from 3-->0 since the module order for our custom kinematics is from 0-->3

They need to be 0->3 everywhere. Specifically, matching the order that you first create your kinematics with, which is FL, FR, BL, BR [here](https://github.com/frc5417/FRC2024-5788/blob/81c50f3caa94b9a4870cca412c913cc54292030b/src/main/java/frc/robot/subsystems/DriveBase.java#L48), which I'm assuming is your 0->3 order.

DragonflyRobotics, 2024-02-09:
> I'm not sure if that's what you were talking about when you said "when i set a starting position".

So I used to set the starting position to something like (2,7) when I constructed odometry but I removed it following the example code and instead checked the "Preset Starting Pose" in PathPlanner GUI to (2,7) but that would cause it to zoom to outer space many times.

DragonflyRobotics, 2024-02-09:
> They need to be 0->3 everywhere. Specifically, matching the order that you first create your kinematics with, which is FL, FR, BL, BR [...]

So I checked the code, and here is how `moduleGroup` corresponds to module location:
* 0 --> FL
* 1 --> FR
* 2 --> BL
* 3 --> BR

Here is how I know: [here](https://github.com/frc5417/FRC2024-5788/blob/81c50f3caa94b9a4870cca412c913cc54292030b/src/main/java/frc/robot/subsystems/Kinematics.java#L38C1-L41C92)

This should make a diamond shape for rotating with 0rad being right facing and the wheel only spins in the forward direction(the direction of the vector).

So I need to ensure that my odom also goes from 0-->3 right?

mjansen4857, 2024-02-09:
Yes, everywhere that you are doing any sort of module ordering it needs to be from 0->3 so that everything matches.

DragonflyRobotics, 2024-02-09:
@mjansen4857 We changed all of our module orders and every single time, the robot followed the path shown on PathPlanner telemetry. The problem still remains that it selects a random path outside the field and goes out forever. Since the robot is consistently following the path shown on telemetry, I seem to believe there might be a deeper rooted issue with our Path generation or execution. What else could we check?

DragonflyRobotics, 2024-02-11:
@mjansen4857 Good news! We did end up fixing the issue but we are not entirely sure why it works. We merely set the robotRelativeChassis speed function to return a blank ChassisSpeed and almost everything seems to function. I can imagine that maybe something is being sent wrong or parsed wrong but do you know of any better explanation? It would be better to go to competition knowing the cause and the reason the solution works the way it does.

mjansen4857, 2024-02-12:
So you are definitely returning chassis speeds wrong. You might have an infinite or NaN value or something potentially. I wouldn't recommend leaving that at 0 since it would break replanning a little bit if you end up needing it.

DragonflyRobotics, 2024-02-12:
What specifically would it break if it is a feature we don't plan on using, we might not look into it. Also, would the rest of the functionality work just fine?

closed 2024-09-13 by mjansen4857