mjansen4857/pathplanner #609: Add an Either Command to the pathplanner events

mmilunicmobile, 2024-02-17:
Something that my team is working on is conditional pathing based on whether we have a game piece in the robot. We are currently making named commands that follow one sequence of paths if we have a piece and a different set if we do not see a piece. This saves time not having to go back.

A solution to this would be including an "either" group in addition to the current "sequential", "parallel", etc. groups. This could allow for much more dynamic branching in auto. Boolean suppliers could be registered with the named commands static class as well.

NickAnastasoff, 2024-03-21:
id love to see this! i came across conditional commands which could help https://github.wpilib.org/allwpilib/docs/release/java/edu/wpi/first/wpilibj2/command/ConditionalCommand.html

mmilunicmobile, 2024-03-22:
After thinking about this for a bit, a hack that you could kinda run would be using a sequential command group inside of a race command group with one other command. The other command could be something like "hasPiece", and when the command starts, it could immediately check if there is a piece, and either end immediately if there is no piece, or continue forever if there is one. This has the effect of the items in the sequential command group only running if there is a piece. You could then make a "!hasPiece" that does the opposite and put that in a race sequence immediately after. This would give you an equivalent to the Either. Since very very few teams would ever use this, and adding the ability to register boolean suppliers is a huge undertaking, I think that this solution would work for now, and we're probably gonna try and see if this works.

mmilunicmobile, 2024-03-22:
This is an example of the sort of composition @NickAnastasoff:

[image]

NickAnastasoff, 2024-03-28:
I think at this point it would make more sense to just create the autos in code with the PathPlanner.fromPathFile function, although that is a cool way to do it!