mjansen4857/pathplanner #610: Mac version crashes immediately

hschilling, 2024-02-19:
**Describe the bug**

When I download the latest version  of the app, 2024.1.4, and launch on my Mac, it dies immediately. This is in the Console. See attached for the full log

```
Translated Report (Full Report Below)
-------------------------------------

Process:               AdvantageScope [37835]
Path:                  /Applications/AdvantageScope.app/Contents/MacOS/AdvantageScope
Identifier:            org.littletonrobotics.advantagescope
Version:               3.0.2 (3.0.2)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-02-19 17:04:36.0303 -0500
OS Version:            macOS 12.1 (21C52)
Report Version:        12
Bridge OS Version:     3.0 (14Y910)
Anonymous UUID:        9FBBA96D-1FA4-6551-6E7C-631BAC9C95EA

Sleep/Wake UUID:       6B24595F-CB5A-419D-8F94-8651A398C97A

Time Awake Since Boot: 84000 seconds
Time Since Wake:       11720 seconds

System Integrity Protection: enabled

Crashed Thread:        0  CrBrowserMain  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       UNKNOWN_0xD at 0x0000000000000000
Exception Codes:       0x000000000000000d, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [37835]

VM Region Info: 0 is not in any region.  Bytes before following region: 4383186944
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      105422000-105424000    [    8K] r-x/r-x SM=COW  ...dvantageScope

Application Specific Information:
objc_msgSend() selector name: occlusionState

Thread 0 Crashed:: CrBrowserMain Dispatch queue: com.apple.main-thread
0   libobjc.A.dylib               	    0x7ff81c17b057 objc_msgSend + 23
1   Electron Framework            	       0x1142953d5 v8::internal::interpreter::BytecodeArrayBuilder::Local(int) const + 147093
2   Electron Framework            	       0x114294f27 v8::internal::interpreter::BytecodeArrayBuilder::Local(int) const + 145895
3   Electron Framework            	       0x114224214 v8::internal::CallInterfaceDescriptor::IsValidFloatParameterRegister(v8::internal::Register) + 4372
4   AppKit                        	    0x7ff81f1c3f74 -[NSApplication _terminateFromSender:askIfShouldTerminate:saveWindows:] + 126
5   AppKit                        	    0x7ff81f1c3e7e __52-[NSApplication(NSAppleEventHandling) _handleAEQuit]_block_invoke + 46
6   AppKit                        	    0x7ff81f212ff8 ___NSMainRunLoopPerformBlockInModes_block_invoke + 25
7   CoreFoundation                	    0x7ff81c3ae4a3 __CFRUNLOOP_IS_CALLING_OUT_TO_A_BLOCK__ + 12
8   CoreFoundation                	    0x7ff81c3ae353 __CFRunLoopDoBlocks + 445
9   CoreFoundation                	    0x7ff81c3ad00d __CFRunLoopRun + 879
10  CoreFoundation                	    0x7ff81c3ac5dd CFRunLoopRunSpecific + 563
11  HIToolbox                     	    0x7ff824fdf4f1 RunCurrentEventLoopInMode + 292
12  HIToolbox                     	    0x7ff824fdf118 ReceiveNextEventCommon + 284
13  HIToolbox                     	    0x7ff824fdefe5 _BlockUntilNextEventMatchingListInModeWithFilter + 70
14  AppKit                        	    0x7ff81eddbb4c _DPSNextEvent + 886
15  AppKit                        	    0x7ff81edda1b8 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1411
16  AppKit                        	    0x7ff81edcc5a9 -[NSApplication run] + 586
17  Electron Framework            	       0x1142e5416 v8::internal::compiler::CodeAssemblerVariable::IsBound() const + 8566
18  Electron Framework            	       0x1142e35a1 v8::internal::compiler::CodeAssemblerVariable::IsBound() const + 769
19  Electron Framework            	       0x1175f1fdd node::AsyncResource::get_async_id() const + 9572109
20  Electron Framework            	       0x1175b793f node::AsyncResource::get_async_id() const + 9332847
21  Electron Framework            	       0x11662afe0 v8::internal::SetupIsolateDelegate::SetupBuiltins(v8::internal::Isolate*, bool) + 4122592
22  Electron Framework            	       0x11662c992 v8::internal::SetupIsolateDelegate::SetupBuiltins(v8::internal::Isolate*, bool) + 4129170
23  Electron Framework            	       0x11662893f v8::internal::SetupIsolateDelegate::SetupBuiltins(v8::internal::Isolate*, bool) + 4112703
24  Electron Framework            	       0x114404c0d v8::internal::compiler::BasicBlock::set_loop_header(v8::internal::compiler::BasicBlock*) + 17053
25  Electron Framework            	       0x114405c57 v8::internal::compiler::BasicBlock::set_loop_header(v8::internal::compiler::BasicBlock*) + 21223
26  Electron Framework            	       0x114405a6f v8::internal::compiler::BasicBlock::set_loop_header(v8::internal::compiler::BasicBlock*) + 20735
27  Electron Framework            	       0x114404505 v8::internal::compiler::BasicBlock::set_loop_header(v8::internal::compiler::BasicBlock*) + 15253
28  Electron Framework            	       0x114404607 v8::internal::compiler::BasicBlock::set_loop_header(v8::internal::compiler::BasicBlock*) + 15511
29  Electron Framework            	       0x1140c982b ElectronMain + 155
30  dyld                          	       0x113bed4fe start + 462
```

If I try installing from the App Store I get "Unable to download app. PathPlanner app could not be installed. Please try again later"
[PathPlanner crash log.txt](https://github.com/mjansen4857/pathplanner/files/14337576/PathPlanner.crash.log.txt)

**To Reproduce**

Download zip file, unzip, launch. 

**Expected behavior**
The app should launch

**Screenshots**
NA

**Versions: (please complete the following information):**
 - OS: MacOS 12.1
 - GUI Version: 2024.1.4
 - PPLib Version: NA
 - PPLib Language: NA

**Additional context**
Add any other context about the problem here.

mjansen4857, 2024-02-23:
`defaults delete com.mjansen4857.pathplanner` in terminal. It seems to be a crash within a library that I can’t really fix at this point.

mjansen4857, 2024-02-23:
Also the App Store version won’t install because you need to delete the manual install version first if they are the same version number.

hschilling, 2024-02-26:
Thanks for the replies! 

It magically started working. I don't think I did anything but obviously something changed. 

Good to know your tips on this though for future use.

Really love PathPlanner so thanks for your work on this!

closed 2024-02-27 by mjansen4857