mjansen4857/pathplanner #614: Robot heading flipped in teleop

HolyArtist31656, 2024-02-25:
**Describe the bug**
When running autos on the Red alliance side, the auto paths are properly flipped and everything runs smoothly, but when the drivers take control when teleop starts, the heading seems to be flipped, i.e. what the code considers "forward" is the opposite direction as it should be. I might just be I implementing something in a wierd way, but many other teams have reported the same issue.

**Versions: (please complete the following information):**
 - OS: [Windows 11]
 - GUI Version: [latest as of submission]
 - PPLib Version: [latest as of submission]
 - PPLib Language: [Java]

bruingineer, 2024-02-25:
I’m wondering if we need to invert the joysticks based on what alliance we are on?

HolyArtist31656, 2024-02-25:
> I’m wondering if we need to invert the joysticks based on what alliance we are on?

So that was what I had done as a "bandaid" solution, and it fixed the issue for the drivers, but I'm not sure if that's the appropriate solution for downstream applications, i.e. I think this results in the "forward" heading for the blue alliance to be 0 and the red alliance to be 180. I'm not sure if that will mess with downstream applications of tracking pose.

bruingineer, 2024-02-25:
I believe that is correct since PathPlanner always uses the blue side as (0,0).

HolyArtist31656, 2024-02-27:
Sounds good

HolyArtist31656, 2024-02-27:
> I believe that is correct since PathPlanner always uses the blue side as (0,0).

Sounds good.

KeremAlperBatal, 2024-03-01:
i have the same problem. Did you solve it?

mjansen4857, 2024-03-03:
The field origin will always remain on the blue side of the field. For field relative drive, you need to adjust the field relative speeds accordingly when on the red alliance. For example, +X velocity will now be towards the red alliance wall. You need to negate your X and Y field relative velocities when on red (only for driving with joysticks).

closed 2024-03-03 by mjansen4857