mjansen4857/pathplanner #624: Erratic Rotation Regardless of Path or Override

nisunku, 2024-03-03:
**Describe the bug**
We've set several autonomous routines using the PathPlanner UI, each comprised of several paths. In each path, we've set rotation targets and the robot rotation and heading of each waypoint. However, no matter what we set, while the translation of the robot behaves as expected, it rotates completely erratically, with no observable relationship to the expected rotation.

We have even tried overriding the rotation using the Holonomic Controller static method as stated in the documentation to no avail.

Lastly, we tried setting the rotation PID values to zero inside the AutoBuilder ConfigureHolonomic method in the drive subsystem constructor.  The robot continued to rotate in an unpredictable way.  This confused us the most.

**To Reproduce**
Steps to reproduce the behavior:
1. Use the code provided in the FRC 2024 file in the Penguin Empire repository (linked below)
- https://github.com/PenguinEmpire/FRC2024
2. Use the routine w/paths shown below
3. Robot rotation behaves unpredictably 

**Expected behavior**
The robot rotates according to the set rotation targets and angles in the path.

**Screenshots**
[image]
[image]
[image]
[image]

**Versions: (please complete the following information):**
 - OS: MacOS 12.7
 - GUI Version: 2024.1.2
 - PPLib Version: 2024.1.2
 - PPLib Language: Java

We would greatly appreciate any help with this; we have tried everything we can think of, but nothing seems to fix it.

bruingineer, 2024-03-04:
What does the `0.15` in the auto builder mean? I believe this argument is for the maximum radius from the center of the robot to where the wheel touches the ground (looking from the top down). 

Looking at your wheel base and track width, this should be around `sqrt(15*15+11.58*11.58) inches to meters = 0.4813`, plus or minus some inches depending on which swerve modules you’re using.

nisunku, 2024-03-04:
Thank you for noticing this! However, we think that this won't affect the rotation. 

Update: We have switched to making our own MoveCommand for autonomous, so if you are looking at the current code from the repo, there won't be any PathPlanner code. I'm providing the link to the commit when the PathPlanner code was still there, so if anyone would like to look at it, please feel free (we still want to use this in the future for our second comp)!

[https://github.com/PenguinEmpire/FRC2024/commit/1acbb15a72f33aa06e5886e76a70f6d4edef420b](url)

bruingineer, 2024-03-04:
This will definitely affect rotation. I know because we made the same change, but in the other direction. We set the max radius to a way to big number because of a unit conversion, and the robot barely would rotate. 

In this case, you have a smaller number set, so it would seem that it’s cause much larger rotations than needed. And then the rotation PID is trying to correct causing oscillations. 

I would recommend making that number correct and run the PathPlanner auto again.

nisunku, 2024-03-04:
Oh, thank you for letting us know! We will definitely play around with this at our next meeting, although we still think there is something wrong with our odometry.

closed 2024-09-13 by mjansen4857