mjansen4857/pathplanner #630: C++ Desktop Build Linking Fails

MrRedness, 2024-03-09:
**Describe the bug**
When building a 2024 wpilib C++ codebase that has pathplannerlib 2024.2.5 installed, if desktop/simulation support is enabled, linking fails. (worked fine on 2024.2.3).

**To Reproduce**
Steps to reproduce the behavior:
1. Install pathplannerlib 2024.2.5 as vendor library
2. Enable desktop support
3. Build or run simulation
4. See error

**Expected behavior**
No error?

**Screenshots**
[image]

**Versions: (please complete the following information):**
 - OS: Windows 10
 - GUI Version: N/A
 - PPLib Version: 2024.2.5
 - PPLib Language: C++

MrRedness, 2024-03-09:
Downloading and copying pathplannerlib into the project seems to work without issue. Unsure if this means the solution is related to using it as a vendor library or it was fixed in an update that hasn't been uploaded yet.

mjansen4857, 2024-03-10:
Is your WPILib version updated? It is built against the latest wpilib

BarrelDev, 2024-04-07:
Have you tried updating your Visual Studio Build Tools? Our team was facing the same issue but that resolved it.

closed 2024-09-13 by mjansen4857