mjansen4857/pathplanner #638: Scrollable path & auto lists

jonahsnider, 2024-03-12:
**Is your feature request related to a problem? Please describe.**

We have A Lot of paths & autos in our PathPlanner project this year. So many that on some screen sizes, the list of paths/autos renders off the screen, making them inaccessible for editing.

**Describe the solution you'd like**

The sections in the project browser for paths & autos should be scrollable, so that all the items can be viewed.

**Additional context**
N/A

mjansen4857, 2024-03-12:
Could you send some screenshots of what it looks like? The lists are scrollable so I’m not sure why it’s not working for you.

jonahsnider, 2024-03-12:
Seems like the paths/autos themselves are scrollable, but folders are not:

[image]

mjansen4857, 2024-03-12:
Yeah I don't think folders are included in the part that's scrollable. Should hopefully be an easy fix

closed 2024-03-14 by mjansen4857