mjansen4857/pathplanner #643: Second Race Group in Auto sequence appears to run incorrect Named Command

gearsincorg, 2024-03-13:
**Describe the bug**
I am using a Race Group in an auto sequence to end a path prematurely.  ( when it sees a Note.)
I have two different commands that I use to race with a path depending on what stage of the Auto sequence I'm in.

My problem occurs when I have two Race Groups in an Auto sequence  see below.

[image]

The first one terminates on an immediate detection, and the second is meant to wait for a triggering event before looking for notes.

I have spend several days trying to find out why the second note detection was not postponing it's detection.  It would work fine when it was the first Race Group in the Auto Sequence, but not the second.  I assumed I had some initialization error.

I initially had one Java command with different constructor parameters, but eventually I just created two independant java commands to verify that it wasn't some naming issue or other shared data issue.  

BOTTOM LINE ....I have been able to demonstrate that instead of the correct named commaned being run in the second Race Group, the same command from the first group is being run again.   This would explain why a single use of this approach works.  I have not been able to come up with any reason why this should be the case...  

This duplicate execution is despite the fact that the two named commands are running two different Java commands .
Also, the correct path is run in both Race Groups, so these are not being repeated.

[image]

I have attempted multiple ways to prove that it was my software acting up here, but I can't.  I have included my Path file as a txt file to see if there is something obvious in there.  Any insight would be appreciated....

[A-1-2-5.auto.txt](https://github.com/mjansen4857/pathplanner/files/14591819/A-1-2-5.auto.txt)

**Versions: (please complete the following information):**
 - OS:  Windows 10
 - GUI Version:  2024.1.4
 - PPLib Version: 2024.2.5
 - PPLib Language:  Java

mjansen4857, 2024-03-13:
How have you demonstrated that it is running the same command? There really isn't a way for it to pull the wrong command since it just uses the name in the file as a key for a hash map. My random guess would be that you are registering `LateStart` again somewhere, which overrides the definition you think it has? 

Not sure what you've tried, but I would try changing your `FindNote` and `LateStart` named commands to be `Commands.print("FindNote")` and `Commands.print("LateStart")` respectively. If these both print correctly, the issue is likely in your late start command causing it to look like your find note command? If late start doesn't print, its probably being overriden by another command somewhere. If find note prints twice, I would be very, very confused.

GaCo1629, 2024-03-13:
My two commands print two different messages to the Dashboard.  I made each command print incrementing numbers each execute call and I see the same command start at 1 each time  eg: Find 1, find 2, find 3   in both cases, whereas the second Race group should be printing  Find late 1, find late 2, find late 3 etc.

As I mentioned, I created an auto with just the one race group.  I can switch the named command and it runs which ever one I include (looking at the message on the dashboard..   So I know both are registered only once.

It really is quite mystifying.

mjansen4857, 2024-03-14:
Could you still try replacing the commands being registered with print commands just to isolate the problem? Also a link to your code, if possible, would be helpful.

gearsincorg, 2024-03-14:
I'm sure this is a dumb question, but where does Commands.print() print to?

mjansen4857, 2024-03-15:
The console you can see in the driver station or riolog in VSCode