mjansen4857/pathplanner #662: Control Length of 0 distance becomes NaN (Breaks & Crashes GUI)

Derp88, 2024-03-25:
**Describe the bug**
The point heading and control length can be come become NaN (Not A Number) if the control point is dragged on top of the anchor point (Control Length is then 0). Once the value becomes NaN, it cannot be changed back to a proper number.

**To Reproduce**
1. Drag a control point to be exactly on top an anchor point (`Next Control Length` = 0)
2. Try to change the the current `Next Control Length` to a number besides 0 by typing in a new number
3. `Next Control Length` will now become NaN
4. It is now impossible to change the value from NaN and if you attempt to do so the generated robot path disappears.
5. If you leave and try to re-open the path, PathPlanner GUI will crash.

**Expected behavior**
- The value should not become NaN when changed from 0 to another number
- Maybe have a way to separate overlapping anchor and control points?

**Screenshots**
Setting the control point on top of the anchor point (by dragging it)
[image]
Trying to change length to a non-zero value (1 in this case), it instead becomes NaN
[image]
`Simulated Driving Time` and the time on the time bar also become NaN
[image]

**Versions:**
 - OS: Windows 10
 - GUI Version: 2024.1.6
 - PPLib Version: 2024.2.7
 - PPLib Language: Java

**Additional context**
This happened because I accidently put the control point on top of the anchor point. I couldn't figure out any way to move the control point besides changing the value for length as you cannot drag it off if the 2 points overlap.

ModBoyex, 2024-04-01:
Yes, I ran into this too because I was trying to make a python script that would write some autos, and it did this exact thing by accident and there no way to move it off.

mjansen4857, 2024-10-02:
Fixed in #774

closed 2024-10-02 by mjansen4857