mjansen4857/pathplanner #667: "Decorate" the robot rectangle

pordonj, 2024-03-26:
**Is your feature request related to a problem? Please describe.**
When drawing paths I'd like to visualize different simple things about my robot (intake region, shot path). I can imagine/eyeball/straight-edge these, but decorating my robot rectangle would offload a lot of this effort.

**Describe the solution you'd like**
Some way in the overall config to apply decorations to the robot rectangle on a given side (for simplicity, I think these could be centered with a given width/depth from the center of one of the robot sides).

**Additional context**
(Bad) Artist rendition of my idea here (colors/thicknesses not to scale)
[image]

I am totally down to do the work for this in a PR, but I poked around for a short time and couldn't figure out where/how the robot rectangle gets drawn

ModBoyex, 2024-04-01:
Personally I would do some of this using mechanisms in the glass simulations. I could see this getting kinda messy.

mjansen4857, 2024-11-07:
Depends on #902 as this can be an addition to that configurator

closed 2024-11-15 by mjansen4857