mjansen4857/pathplanner #677: Linker Error when using PPLib 2024.2.8

ejbunnell, 2024-04-04:
**Describe the bug**
When I update my PPLib to version 2024.2.8 from 2024.2.3, it creates a Linker Error when I build the C++ code. 

**To Reproduce**
Steps to reproduce the behavior:
1. Update PPLib to version 2024.2.8
2. Compile C++ code
3. See many errors that are not present when PPLib is in version 2024.2.3

**Expected behavior**
The build succeeds in 2024.2.3. I am not sure if there are newer versions before 2024.2.8, but I know 2024.2.3 works.

**Screenshots**
[image]

**Versions:**
 - OS: Windows 11
 - GUI Version: 2024.1.6
 - PPLib Version: 2024.2.8
 - PPLib Language: C++

mjansen4857, 2024-04-04:
Is your WPILib version up to date?

ejbunnell, 2024-04-04:
It was not, but now it is. The error is still present. I believe it might stem from this: 

PathplannerLib.lib(LocalADStar.obj) : error LNK2019: unresolved external symbol _Thrd_sleep_for referenced in function "void __cdecl std::this_thread::sleep_until<struct std::chrono::steady_clock,class std::chrono::duration<__int64,struct std::ratio<1,1000000000> > >(class std::chrono::time_point<struct std::chrono::steady_clock,class std::chrono::duration<__int64,struct std::ratio<1,1000000000> > > const &)" (??$sleep_until@Usteady_clock@chrono@std@@V?$duration@_JU?$ratio@$00$0DLJKMKAA@@std@@@23@@this_thread@std@@YAXAEBV?$time_point@Usteady_clock@chrono@std@@V?$duration@_JU?$ratio@$00$0DLJKMKAA@@std@@@23@@chrono@1@@Z)

I'm not entirely sure, however.

BarrelDev, 2024-04-07:
Have you tried updating your Visual Studio Build Tools? Our team was facing the same issue but that resolved it.

ejbunnell, 2024-04-08:
That seems to have fixed the issue. Thank you!