mjansen4857/pathplanner #721: Support branching autos from the UI

lidor51, 2024-09-14:
Defining autonomous as a prioritized decision tree, each node can be either a path or a decision tree of its own.
Doing this from the UI would allow user-friendly way to define complex autos.

[image]

Each branch would have a condition, conditions would be checked from left to right (this is why it's a "prioritized" decision tree). We would like to make the paths (\w their conditions?) draggable.
Conditions should be reusable for a user-friendly experience.
If neither of the conditions are met, the auto finished.

lidor51, 2024-09-14:
I suggest a vertical scrolling in this window, and scrolling/right-left buttons horizontally.
Alongside keyboards arrows navigation in the tree.

mjansen4857, 2024-09-25:
I think this would be useful, but this is a very large and time-consuming feature to implement. So, I'm not currently targeting it for 2025 unless all remaining tasks for 2025 are done with enough time left to properly implement it before the end of beta.

lidor51, 2024-09-25:
Indeed a big feature, I suggest to keep it open to a future iteration.

HenryLeC, 2024-12-07:
This isn't exactly what you were thinking but this is achievable with Conditional Command groups #923 
@lidor51 Is there anything you can think of that a directed graph could achieve that my pull request can't besides being cleaner?

lidor51, 2024-12-07:
Hey that's cool! Besides the different visualisation, it seems like there're a few differences from my suggestion.

I have a few questions on the screenshot you put:
[image]

1. Is 'Shoot from 1' is "on true" & 'Sweep line' is "on false"?
If so, not sure if this is a self explainatory experience.

2. Is it possible to add conditional group inside another conditional group's onTrue/onFalse?

3. IMO it's a very important feature being able to save a named sub-branch so you don't need to duplicate and re-define it again in every auto. Alternatively, a more general approach is supporting "sub-autos" (practically, being able to put an auto as a component of another auto).

HenryLeC, 2024-12-07:
1. Yes, thats exactly what it is. It's the same format as the code, and it also explains it in a tooltip if you hover over the group name. The docs will also be helpful for this
2. Yes, that is possible. You would just need to add a conditional command group into the group for onTrue or onFalse. (A conditional command group is not a command group in the pathplanner UI)
3. This should be very easy to implement and I am willing to do sub autos if @mjansen4857 will allow it

lidor51, 2024-12-09:
1. In both conditional group & deadline, I believe the appropriate name should be "conditionName" rather than "conditionalName".
2. Cool.
3. I think it's very important not only in this feature context, autos duplication is very annoying and requires a lot of redundant maintenance.