mjansen4857/pathplanner #723: Sorting Method not Being Saved

JosephTLockwood, 2024-09-20:
**Describe the bug**
After applying a sort, changing a setting, and going into and out of a path, the sort method is not saved.

**To Reproduce**
Steps to reproduce the behavior:
1. Apply a sort method to paths
2. Go into settings
3. Exit and go back into a path and exit
4. Sorting should no longer be applied.

**Expected behavior**
The sort method should be saved and applied until it is changed.

mjansen4857, 2024-09-24:
I can't seem to reproduce this. Could you give more info on what build you're using and perhaps a screen recording of this behavior?

JosephTLockwood, 2024-09-25:
I was using PathPlanner-Windows-v0.0.0-dev-871.
[Link](https://drive.google.com/file/d/13Ps4NhljGf7UwJf6uZA7I0f4vl588zc1/view?usp=sharing) to video of the issue.

mjansen4857, 2024-09-25:
Ok so you're using a build from #705 meaning its a bug with that PR @WindingMotor

WindingMotor, 2024-09-25:
Very strange Ill take a look at it

closed 2024-09-30 by JosephTLockwood