mjansen4857/pathplanner #780: the path ignores the acceleration limit if the last and next segments converge

jizelda, 2024-10-01:
**Describe the bug**
if the robot needs to reverse it's direction at a specific point, such that the last segment and the next segment converge, it will use it's won't decelerate, and instead use its' last velocity to turn backward, effectively using infinite acceleration.

[image]
in this image you can see that the points remain green, even though the robot should probably decelerate and then reaccelerate in the opposite direction. 

The folder of the path pictured:
[pathplanner.zip](https://github.com/user-attachments/files/17212270/pathplanner.zip)

**To Reproduce**
Steps to reproduce the behavior:
-Try to align the control points such that the previous and the next parts of the path converge. 
-It might also happen when optimizing crazy paths.

**Expected behavior**
A clear and concise description of what you expected to happen.
I expect the robot to decelerate and then re cerate in the opposite direction

**Versions: (please complete the following information):**
 - OS: Windows 10
 - GUI Version: the latest version of main, compiled manually. latest commit is 997a2bd
 - PPLib Version: the latest version of main. latest commit is 997a2bd
 - PPLib Language: java

**Additional context**
Add any other context about the problem here.
This problem is really avoidable while making paths, but because it allows for really fast acceleration, the optimizer uses it to "cheat" on sharp turns.

For example, here is a optimized path with 4 "cheats" in a row
[image]
The the path pictures
[New Path.zip](https://github.com/user-attachments/files/17212423/New.Path.zip)

mjansen4857, 2024-10-01:
The generation is not meant to handle path discontinuities, which placing control points like that creates. If you want a "reversal" effect like that, you need to create multiple paths. Anywhere that the robot would need to come to a complete stop must be split into different paths.

Supporting discontinuities may be a future update but is not currently planned.

mjansen4857, 2024-10-01:
True, but I wouldn't really trust the optimizer to give perfect results all the time anyways. If there's some weird placement of control points that finds an issue in the generator like that, it will very likely find it. It will always need to be sort of "supervised" to make sure the optimized path doesn't do something weird.

I will look into discontinuities at some point, there's a chance I could just insert a path point at those waypoints with a max velocity of 0 and have it work, we'll see. There's quite a few more important things I need to finish before that though.