mjansen4857/pathplanner #831: Changing the "Wheel Redius" cause the path generator to fail

Ramtech59Sensei, 2024-10-07:
**Describe the bug**
changing the value in the field "Wheel Redius" cause the path editor or the auto path editor to fail 

**To Reproduce**
from this 
[image]
to this 
[image]
cause this
[image]

Ramtech59Sensei, 2024-10-07:
[test.txt](https://github.com/user-attachments/files/17271707/test.txt)
path example , it happens with every path
if I change the wheel radius back to the original value all work again

Ramtech59Sensei, 2024-10-07:
also happens if you change only the value for "True Max Max speed" and "Max Velocity" to 4.1 but those not happens if you change it to 4.3

mjansen4857, 2024-10-07:
Your true max speed is very, very slow for the rest of your setup. The gear ratio is 6.12 which is the same as an SDS L3 module. The true max speed for that ratio is closer to 5 m/s. Because you have it so low, it is assuming that it takes a lot of effort to go that speed since the motor can provide a lot of torque at that speed. That leaves 0 torque left over to accelerate the robot without violating the current limit, which is why it can't generate.

Ramtech59Sensei, 2024-10-07:
Can you put a label or labels similar to "Max optimal Limit" to show calculated torque value.
Show in red when is incorrect or will not generate path ?

Jon-Zimmerman, 2024-10-07:
> Your true max speed is very, very slow for the rest of your setup. The gear ratio is 6.12 which is the same as an SDS L3 module. The true max speed for that module is closer to 5 m/s. Because you have it so low, it is assuming that it takes a lot of effort to go that speed since the motor can provide a lot of torque at that speed. That leaves 0 torque left over to accelerate the robot without violating the current limit, which is why it can't generate.

This might be unintuitive to the user as the resulting behavior is not explained as a result of wrongly entered max speed. To get the correct behavior it sounds like you would require a manual recalculation of the correct max speed with every adjustment of the wheel radius and gearing. 
Choreo handles this by providing a theoretical calculation of the max floor speed based on the motor parameters, gearing and wheel radius. Additionally Choreo uses a user entry for max motor rpm to make this calculation. This allows the user to choose a number with some overhead to avoid saturating the controller (running at max motor rpm gives no room for the controller to catch up if it falls behind the desired state.)