mjansen4857/pathplanner #835: startingStateJson is null

AydenLo17, 2024-10-07:
﻿﻿﻿﻿﻿﻿﻿﻿ERROR ﻿﻿ 1 ﻿﻿ Unhandled exception: java.lang.NullPointerException: Cannot invoke "org.json.simple.JSONObject.get(Object)" because "startingStateJson" is null ﻿﻿ com.pathplanner.lib.path.IdealStartingState.fromJson(IdealStartingState.java:21) ﻿﻿﻿
﻿﻿﻿﻿﻿﻿ Error at com.pathplanner.lib.path.IdealStartingState.fromJson(IdealStartingState.java:21): Unhandled exception: java.lang.NullPointerException: Cannot invoke "org.json.simple.JSONObject.get(Object)" because "startingStateJson" is null ﻿
﻿﻿﻿﻿﻿﻿﻿﻿Warning ﻿﻿ 1 ﻿﻿ The robot program quit unexpectedly. This is usually due to a code error.
  The above stacktrace can help determine where the error occurred.
  See https://wpilib.org/stacktrace for more information. ﻿﻿ edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:379) ﻿﻿﻿
﻿﻿﻿﻿﻿﻿ 	at com.pathplanner.lib.path.IdealStartingState.fromJson(IdealStartingState.java:21) ﻿
﻿﻿﻿﻿﻿﻿﻿﻿ERROR ﻿﻿ 1 ﻿﻿ The startCompetition() method (or methods called by it) should have handled the exception above. ﻿﻿ edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:386) ﻿﻿﻿
﻿﻿﻿﻿﻿﻿ 	at com.pathplanner.lib.path.PathPlannerPath.fromJson(PathPlannerPath.java:547) ﻿
﻿﻿﻿﻿﻿﻿ 	at com.pathplanner.lib.path.PathPlannerPath.fromPathFile(PathPlannerPath.java:316) ﻿
﻿﻿﻿﻿﻿﻿ 	at com.pathplanner.lib.auto.CommandUtil.pathCommandFromData(CommandUtil.java:72) ﻿
﻿﻿﻿﻿﻿﻿ 	at com.pathplanner.lib.auto.CommandUtil.commandFromJson(CommandUtil.java:46) ﻿
﻿﻿﻿﻿﻿﻿ 	at com.pathplanner.lib.auto.CommandUtil.sequentialGroupFromData(CommandUtil.java:80) ﻿
﻿﻿﻿﻿﻿﻿ 	at com.pathplanner.lib.auto.CommandUtil.commandFromJson(CommandUtil.java:47) ﻿
﻿﻿﻿﻿﻿﻿ 	at com.pathplanner.lib.commands.PathPlannerAuto.initFromJson(PathPlannerAuto.java:134) ﻿
﻿﻿﻿﻿﻿﻿ 	at com.pathplanner.lib.commands.PathPlannerAuto.<init>(PathPlannerAuto.java:57) ﻿
﻿﻿﻿﻿﻿﻿ 	at com.pathplanner.lib.auto.AutoBuilder.buildAutoChooserWithOptionsModifier(AutoBuilder.java:374) ﻿
﻿﻿﻿﻿﻿﻿ 	at com.pathplanner.lib.auto.AutoBuilder.buildAutoChooser(AutoBuilder.java:333) ﻿
﻿﻿﻿﻿﻿﻿ 	at com.pathplanner.lib.auto.AutoBuilder.buildAutoChooser(AutoBuilder.java:321) ﻿
﻿﻿﻿﻿﻿﻿ 	at frc.robot.RobotContainer.<init>(RobotContainer.java:349) ﻿
﻿﻿﻿﻿﻿﻿ 	at frc.robot.Robot.robotInit(Robot.java:91) ﻿
﻿﻿﻿﻿﻿﻿ 	at org.littletonrobotics.junction.LoggedRobot.startCompetition(LoggedRobot.java:83) ﻿
﻿﻿﻿﻿﻿﻿ 	at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:365) ﻿
﻿﻿﻿﻿﻿﻿ 	at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:453) ﻿
﻿﻿﻿﻿﻿﻿ 	at frc.robot.Main.main(Main.java:26) ﻿
﻿﻿﻿﻿﻿﻿  ﻿

mjansen4857, 2024-10-07:
Are you trying to load a path created in the 2024 version? PPLib 2025 is not compatible with 2024 paths. You need to re-save them from the 2025 GUI.

AydenLo17, 2024-10-07:
Yes i believe that is the issue, I couldn't get the GUIto automatically resave them so i just deleted them and will remake them. No issues so far.

mjansen4857, 2024-10-07:
If you edit the path and just hit enter in some text field like a waypoint position it will save the file

mjansen4857, 2024-10-11:
Will be notified out outdated files now due to #852

closed 2024-10-11 by mjansen4857