mjansen4857/pathplanner #860: PathPlannerAuto triggers

lidor51, 2024-10-12:
Potentially useful auto triggers are:
Distance* from start 
Distance* from end 
Time left to end

*(can be norm2 distance or trajectory distance)

katzuv, 2024-10-24:
I would like to generalize "time left to end" to time left to event marker – "open the intake 1 second before reaching the intake position".

lidor51, 2024-10-24:
@katzuv not sure that I really got you. All the PathPlannerAuto triggeres @mjansen4857 added are auto/path related.

What's the "intake position" in your example?

Are you suggesting to add timeBefore/AfterTrigger(triggerName) (or NamedCommand)?
If this is case why not to just add another trigger/named command in the path? (Also that doesn't sound like a generalization to what I've suggested, just another trigger suggestion).

katzuv, 2024-10-24:
> why not to just add another trigger/named command in the path?

For when you don't want to use named commands or save on event markers, or for on-the-fly paths.

I see this as a generalization because instead of time to specifically end position, it's time to any other position (or event marker).

lidor51, 2024-10-24:
I understand know, nice idea. Your suggestion actually might generalize all of my suggested triggers.

But, what are the expected params? Pose2d? What if the pose isn't on the trajectory? (One option is to add also distance threshold param, but it might also not be in the threshold as well).
I think its need to be really well defined.

Anyways, regardless of @katzuv suggestions, I still highly suggest to keep also my suggested triggers **as is**, as they're more intuitive and requires no params.

katzuv, 2024-10-29:
Sorry, I missed your comment. I think an event marker should be used because it's guaranteed to be on the path.

mjansen4857, 2024-11-04:
So the original suggestions here are either not possible to implement or pretty redundant.

> Time left to end

Not possible to implement properly since the auto has no idea how long it will actually take to run due to the possibility of inserting commands between paths that take an unknown time to run

> Distance* from start
Distance* from end

Difficult to implement in a useful way since it would need to take all paths in to account. This is also essentially what an event marker is anyways, so its probably best to just use that.

However, I think triggers for time before/after an event marker (perhaps distance as well) could be quite useful.

closed 2024-11-07 by mjansen4857