mjansen4857/pathplanner #910: Specific Paths in Autos Don't Flip

PickleFace5, 2024-11-18:
**Describe the bug**
When running an auto that consists of multiple paths, paths become unflipped. In my scenario, if you want to flip the path when your on the red alliance, the first 1 or 2 paths will be on the red side, but then on the 3rd path, the robot will travel to the opposite side of the field and complete the auto on the wrong side.

The issue seems to happen under a specific criteria I haven't been able to pinpoint. I have included my team's repository in the steps to reproduce, which is the entire project with 3 autos (you can just copy our PP deploy folder, our code is basically the CTRE SwerveWithPathPlanner example):

- WorkingAuto, made up of 2 paths. This path works as expected on both alliances.
- NotWorkingAuto, made up of the same paths as WorkingAuto, but with an additional 3rd path that experiences this issue.
- SinglePathTest, which is the 3rd path from NotWorkingAuto by itself. This auto works on both alliances.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to [my team's repository](https://github.com/SteelRidgeRobotics/2024MetalMelody/tree/ctr-swerve).
2. In simulation, run each auto.
3. See error

**Expected behavior**
Each path in the auto should flip if the specified condition is met (in this case the DS being red).

**Screenshots**
I have attached a video of an AdvantageScope log that logs the current robot pose (red), the desired pose (green), and the current path/trajectory (green). This video showcases each auto in order from top to bottom.

https://github.com/user-attachments/assets/6f3393d8-0274-426e-987a-6d091739bad5

**Versions: (please complete the following information):**
 - OS: Windows 11
 - GUI Version: v2025.0.0-beta-4
 - PPLib Version: 2025.0.0b4
 - PPLib Language: Python

**Additional context**
No PathPlanner warnings or errors appear in the console, however loop time overruns can be occasionally outputted during the NotWorkingAuto before the 3rd path begins.

mjansen4857, 2024-11-21:
> * WorkingAuto, made up of 2 paths. This path works as expected on both alliances.
> * NotWorkingAuto, made up of the same paths as WorkingAuto, but with an additional 3rd path that experiences this [...]

Which autos are these in your deploy folder? I looked at your repo but don't see anything that seems to match up to these names.

mjansen4857, 2024-11-21:
Nevermind I found the issue, I don't need the autos.

closed 2024-11-21 by mjansen4857