mjansen4857/pathplanner #919: Settings file missing   "robotWheelbase" entry

jmullins16, 2024-11-29:
**Describe the bug**
Using PathPlannerLib "2025.0.0-beta-5" and PathPlanner app 2025.0.05 (1165), AutoBuilder failed to configure due to a  missing JSON entry. robotWheelbase entry did not seem to be in the settings file, and when I added it (our robot is square), AutoBuilder configured fine.

Our beta robot project is located here in GitHub:
https://github.com/usfirst-frc-2135/robot2025/tree/main/Comp2024-Imported

The pathPlanner settings file stored in the repo has had this entry pasted in manually to make it work.

We are using the latest WPILib "2025.1.1-beta-1" and CTRE "25.0.0-beta-3". 

**To Reproduce**
Steps to reproduce the behavior:
1. Use our settings file from the repo in the application
2. Re-deploy the settings
3. robotWheelbase is missing from the settings file
4. simulate the robot project, AutoBuilder fails with a stacktrace

**Expected behavior**
AutoBuilder configures correctly.

**Screenshots**
Error at com.pathplanner.lib.config.RobotConfig.fromGUISettings(RobotConfig.java:274): Failed to load PathPlanner config and configure AutoBuilder
        at com.pathplanner.lib.config.RobotConfig.fromGUISettings(RobotConfig.java:274)
        at frc.robot.subsystems.CommandSwerveDrivetrain.configureAutoBuilder(CommandSwerveDrivetrain.java:241)
        at frc.robot.subsystems.CommandSwerveDrivetrain.<init>(CommandSwerveDrivetrain.java:189)
        at frc.robot.generated.TunerConstants.createDrivetrain(TunerConstants.java:188)
        at frc.robot.RobotContainer.<init>(RobotContainer.java:117)
        at frc.robot.Robot.<init>(Robot.java:24)
        at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:301)
        at edu.wpi.first.wpilibj.RobotBase.lambda$startRobot$0(RobotBase.java:418)
        at java.base/java.lang.Thread.run(Thread.java:840)

**Versions: (please complete the following information):**
 - MacOS: 15.1.1 (24B91)
 - GUI Version:2025.0.05 (1165)
 - PPLib Version: 2025.0.0-beta-5
 - PPLib Language: Java

**Additional context**
Add any other context about the problem here.

mjansen4857, 2024-12-02:
This indicates that you’re not actually using PPLib beta 5 since wheelbase is no longer included in setting which beta 5 is updated to not use. Make sure you’re updated. If you are and it’s still not working, try deleting your gradle cache and building again to force it to pull down the new version.

jmullins16, 2024-12-02:
Still not working for me.

I tried to make gradle with --refresh-dependencies which didn't help, and then just deleted the project .gradle folder entirely to make it rebuild.

Still fails with same error. Unless I'm not understanding how to make gradle pull the beta5 new version, I must be missing something else in the process. Is there a way to request the version from PPLib to confirm it?

Here is the vendordeps file for reference:

`{
    "fileName": "PathplannerLib-beta.json",
    "name": "PathplannerLib",
    "version": "2025.0.0-beta-5",
    "uuid": "1b42324f-17c6-4875-8e77-1c312bc8c786",
    "frcYear": "2025",
    "mavenUrls": [
        "https://3015rangerrobotics.github.io/pathplannerlib/repo"
    ],
    "jsonUrl": "https://3015rangerrobotics.github.io/pathplannerlib/PathplannerLib-beta.json",
    "javaDependencies": [
        {
            "groupId": "com.pathplanner.lib",
            "artifactId": "PathplannerLib-java",
            "version": "2025.0.0-beta-5"
        }
    ],
    "jniDependencies": [],
    "cppDependencies": [
        {
            "groupId": "com.pathplanner.lib",
            "artifactId": "PathplannerLib-cpp",
            "version": "2025.0.0-beta-5",
            "libName": "PathplannerLib",
            "headerClassifier": "headers",
            "sharedLibrary": false,
            "skipInvalidPlatforms": true,
            "binaryPlatforms": [
                "windowsx86-64",
                "osxuniversal",
                "linuxx86-64",
                "linuxathena",
                "linuxarm32",
                "linuxarm64"
            ]
        }
    ]
}`