mjansen4857/pathplanner #921: Targetory Failed to Generate

alexanderszura, 2024-12-05:
The error popped up when I was messing around with the new Point Towards Zones.

I'm not sure how to reproduce it, but I'll include the path file.

[image]

**Versions: (please complete the following information):**
 - OS: Windows 10
 - GUI Version: 2025.0.0-beta-5
 - PPLib Version: 25.0.0-beta-3
 - PPLib Language: Java

Here's a JSON version of the path file
[Full Path.json](https://github.com/user-attachments/files/18026835/Full.Path.json)

mjansen4857, 2024-12-06:
You need to fix your end control point. Having it in the opposite direction of travel like that makes a super sharp turn that the generator can't handle.

alexanderszura, 2024-12-06:
Ah got it thanks, is this something that you will plan on telling the user in path planner?: I saw it on another issue but I didn't think it was the same issue.

mjansen4857, 2024-12-26:
I will probably just change the error message to indicate that a failure like this is going to be due to control point placement. Its not really something I could reliably detect/prevent beforehand. Actually fixing the issue would require much higher path resolution at the "kinks" in the path. This would increase the generation time quite a bit, so it might be something that needs to wait for the new control system.

closed 2025-01-01 by mjansen4857