mjansen4857/pathplanner #922: Slow Start to a Tarjectory

alexanderszura, 2024-12-05:
When the path first starts the robot is very slow, almost not moving at all. Then after about half a second continues on. This might be something I'm doing but I can't figure out what's wrong. 

I don't have any constraint zones just a point towards zone.
[image]

 - OS: Windows 10
 - GUI Version: 2025.0.0-beta-5
 - PPLib Version: 25.0.0-beta-3
 - PPLib Language: Java

**Additional context**
Here is the path and my setting files as JSON's
[Full Path.json](https://github.com/user-attachments/files/18026902/Full.Path.json)
[settings.json](https://github.com/user-attachments/files/18026903/settings.json)

bcrteammember, 2025-03-15:
Did you find a solution to this problem? We are experiencing the exact same symptoms.

alexanderszura, 2025-03-17:
@bcrteammember no we did not, this was just  an issue I found with a weird path we never used. Sorry I can't help you