mjansen4857/pathplanner #925: Generate ChassisSpeeds setpoints are not smooth

mpaulonis, 2024-12-11:
**Describe the bug**
When simulating the AdvantageKit talonfx_swerve project and running the included Example Auto, the ChassisSpeeds setpoints generated by PathPlanner are not smooth.  The screenshot from AdvantageScope shows what "not smooth" is.

[image]

The PP telemetry shows the same variation in the actual velocity, although the commanded velocity in telemetry appears to be ramping smoothly.

[image]

The strange chassis velocity setpoints are even more obvious if you change the max velocity and acceleration on the Example Path to 1.0 each and simulate.

[image]

I've seen the same non-smooth ChassisSpeeds setpoints on an actual robot.  It's not just this simulation.  However, it may be easier to track down the problem because it occurs in simulation.

**To Reproduce**
Steps to reproduce the behavior:
1. Get the [AdvantageKit talonfx_swerve project](https://github.com/Mechanical-Advantage/AdvantageKit/tree/main/template_projects/talonfx_swerve)
2. Simulate it on a computer with [WPILib 2025.1.1 beta 2](https://github.com/wpilibsuite/allwpilib/releases/tag/v2025.1.1-beta-2) installed.
3. Select the Example Auto from the auto chooser and change the robot state to Autonomous.
4. Use AdvantageScope to plot ChassisSpeeds setpoints and optionally measured values

**Expected behavior**
ChassisSpeeds setpoints should ramp smoothly throughout the path.  The "hitches" seen in the screenshots should not be present.

**Versions: (please complete the following information):**
 - OS: Windows 11
 - GUI Version: v2025.0.0-beta-5
 - PPLib Version: 2025.0.0-beta-5
 - PPLib Language: Java

mjansen4857, 2024-12-15:
This is normal. Each one of those steps is an individual robot code loop. AdvantageScope does not smooth out graphs, the PP GUI does.

mpaulonis, 2024-12-15:
The issue I see is not the discretization of the signal.  It is the multiple times where the profile increases (or decreases) rapidly, starts to level out, and then increases (or decreases) again.  For example, in the vx profile n the last graph, between about 23.9 seconds and 24.4 seconds, there is a curve to the profile where it appears to be trying to level out around 0.45 or so.   Then it accelerates again heading to around 0.95 at 24.9 seconds.  This should be a straight line from 0 to 0.95.  The very evident discontinuous hitch in the acceleration phase of the trapezoid should not be there.

mjansen4857, 2024-12-15:
I'm assuming you're logging robot relative speeds here. Those don't really mean too much because they can change quite a bit when rotating/changing directions, etc. You're essentially just logging the individual components of a velocity vector which is not easy to make sense of. The PP GUI shows linear velocity, or the magnitude of that vector, which is easier to make sense of and appears smooth.

There's no reason why the individual x/y velocity components need to be smooth in the same way, the movement of the robot as a whole is smooth, as well as each individual module. From looking at those graphs I don't see anything unexpected.

mpaulonis, 2024-12-15:
I'll record the profiles for a straight run (x-only) with no rotation tomorrow and upload the results.   It's not related to an interaction between the x and y components or an interaction between translation and rotation.

mpaulonis, 2024-12-15:
The graph below is from a simulation using the [AdvantageKit talonfx_swerve project](https://github.com/Mechanical-Advantage/AdvantageKit/tree/main/template_projects/talonfx_swerve), running an auto routine comprised of one path that drives from (1.5, 6.5) to (5.5, 6.5), a straight path of 4 meters in the x direction starting and ending at heading 0.0.  The max velocity is 1 m/s.  The max acceleration is 1 m/s/s.

The expected result is a trapezoidal x velocity profile that takes 1 s to ramp from 0 m/s to 1 m/s, cruises at 1 m/s for 3 s, and then takes 1 s to ramp back down to 0 m/s.  The actual result is shown below.

[image]

The robot-relative chassis speed setpoint in the x direction is the yellow line.  This is what PathPlanner is sending to drive the 4 meter auto path in the x direction.  The path has no y component and no rotational component.

mpaulonis, 2024-12-15:
In case it is of any additional value, here is the velocity profile for a straight run from (1.5, 6.5) to (3.5, 6.5) with max velocity set to 0.5 m/s and max acceleration set to 0.5 m/s/s.  The non-trapezoidal nature of the robot-relative chassis speed setpoints (yellow line) is even more apparent with slower velocity and acceleration limits.

[image]

mjansen4857, 2024-12-16:
What are your path following PID gains?

mpaulonis, 2024-12-16:
5.0 for translation and rotation.  Same as the example code on pathplanner.dev and unchanged from the [AdvantageKit talonfx_swerve template project](https://github.com/Mechanical-Advantage/AdvantageKit/tree/main/template_projects/talonfx_swerve).

mjansen4857, 2024-12-16:
Are you using the SwerveSetpointGenerator here? I'm unfamiliar with the AKit template so idk if it does that.

If not, can you try the same run from the last graph with 0 PID gains to see what that looks like?

mpaulonis, 2024-12-16:
I don't think the AKit template is using SwerveSetpointGenerator, but I'll check to make sure.  I'll also run with 0 gain and have that up shortly.

mjansen4857, 2024-12-16:
Yeah so then the speeds being generated here are fine. It looks like what you're seeing is caused by the PID gains. I'm not really sure why exactly that's happening since the gains you used work fine for most robots. I think you should switch to graphing positions now since that is what the PID controllers act on. A graph of the PP target pose entry and you're robot's current pose might give a better idea of what's happening. Those should both be logged under the PathPlanner table already.

mpaulonis, 2024-12-16:
Here is what target and current poses look like as graphed from the PathPlanner table.

This first graph is with gains = 0.  The yellow line is the x component of the target pose.  The cyan line is the x component of the current pose.  The path is 2m in the x direction only.  The discontinuity in the target pose x looks concerning. There is a red line with the current pose y component and a gray line with the target pose y component completely obscured beneath it.  Both are constant at 6.5, as expected with this straight path at constant y.

[image]

Here is how things change if gains = 5.

[image]

and the velocities are shown here for reference.

[image]

The x controller sees the error between target and current in the first second or so of the path and sends faster chassis speed setpoints to close it.  This results in a rate of change of chassis speed that is above the max acceleration.  Once the chassis closes the gap and overshoots, the acceleration slows down to almost zero.  You can see the correspondence between the slope change in the target pose and the discontinuity in the chassis speed setpoint.

mjansen4857, 2024-12-19:
So I did some testing and it looks like what you're seeing is caused almost entirely by driving so slow. This leads to a pretty significant amount of time between trajectory states, which causes a lot of problems with the interpolation between the two closest states. The exact issue is that the interpolation between states is linear. This is fine for the velocity, but the position does not actually change linearly between states when the velocity is also changing. 

Note the S curve for the blue line, which is the robot's position based on the interpolated speeds versus the linear yellow line for the interpolated position:
[image]

This normally isn't really an issue when going at normal speeds, since the short amount of time between states doesn't equate to much of a difference when linearly interpolating the position. However, I did notice this effect to some extent even at normal speeds, so improving the state interpolation should generally improve accuracy across the board. What I'm going to do is just change the pose interpolation to integrate the field speeds instead of just linearly interpolating.

mjansen4857, 2024-12-19:
First image is basically your current setup 0.5/0.5 constraints, kP = 5.0, linear interpolation for poses between states. Second image only changes to using integration for poses between states.

Red = commanded velocity w/ PID feedback
Orange = target velocity from trajectory state
Yellow = target pose from trajectory state
Blue = current robot position

[image]
[image]

mpaulonis, 2024-12-19:
Going from linear interpolation of position to integration of velocities seems like a significant enhancement. The very first graph I posted where velocities max out near 3 m/s shows clear evidence of the linear interpolation side effect.  I'm sure that this change will make a big difference at all velocities. Thanks for your amazing work gifted to all teams.

PR #935 (2024-12-19): Integrate field speeds to interpolate state positions
Fixes #925
closed 2024-12-19 by mjansen4857