mjansen4857/pathplanner #936: Named Commands + EventTriggers breaks autons

CrispyBacon1999, 2024-12-19:
**Describe the bug**
If a single subsystem has an EventTrigger, as well as NamedCommands set up for it, when the robot hits the EventTrigger, it will immediately stop. I have an EventTrigger for revving the flywheel (while following the path and approaching the shoot point), then a NamedCommand for actually running the shooting logic in between path follows.

Ideally, I'd like to be able to run all of my logic with just EventTriggers, but it doesn't appear that making named commands in PathPlanner GUI (in between path following) will trigger those.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to an autonomous mode
2. Add a named command between 2 paths that requires the flywheel
3. Go to a path
4. Add an EventMarker that requires the flywheel (I used a zoned one, not sure if that affects anything)
5. Run the path
6. Robot stops as soon as the EventMarker is hit

**Expected behavior**
Robot should handle both types of event markers together correctly, or PathPlanner should be able to add EventTriggers to autonomous routines.

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]
[image]

**Versions: (please complete the following information):**
 - OS: Windows 11
 - GUI Version: v2025.0.0-beta-5
 - PPLib Version: 2025.0.0-beta-5
 - PPLib Language: Java

mjansen4857, 2024-12-20:
This is not a bug and is expected behavior of the commands framework. Your auto routine will have the requirements of all the commands it contains. If you set up a trigger to run a command that shares requirements, it will cancel the auto command. 

If you want to keep the same structure of mixing triggers and named commands, you will have to run your named commands within a proxy command to avoid the auto routine from inheriting its requirements.

CrispyBacon1999, 2024-12-20:
Proxied commands seems to fix it. Is there a more recommended way to use EventTriggers as blocking commands to prevent the path from being followed other than just using NamedCommands in the auto routine?

mjansen4857, 2024-12-21:
No. EventTriggers can only be triggered while path following, so they can't be run in between paths. Triggers by nature don't really work in a blocking way like that anyways, so it serves a different purpose from traditional command composition.

closed 2024-12-26 by mjansen4857