mjansen4857/pathplanner #965: Wheels not being rendered

me-it-is, 2025-01-04:
The wheels are not being shown.

To make the wheels not show you just have to copy the settings shown in the screenshots.

I have the wheels set to 0.04m radius so because they are only slightly smaller than the default wheel they should be quite large

[image]

[image]

 MacOS 15.1.1
 Gui 2025.0.0 beta 6
 PPLib 2024.2.8
 Java

Im trying to create a profile for a romi
[https://www.pololu.com/product/4022](url)

mjansen4857, 2025-01-04:
The modules are not drawn when holonomic mode is off. Those are only meant to represent swerve modules so this is intentional.

me-it-is, 2025-01-04:
Thanks thats good to know.

closed 2025-01-04 by me-it-is