mjansen4857/pathplanner #966: C++ named command registration segfault (at least on aarch64/x86)

michaelgtodd, 2025-01-04:
Up front disclaimer, this is running on an x86 linux (ubuntu) computer in our ROS ecosystem. I haven't had a chance to try it in a different environment yet, but I wanted to make sure I wasn't doing something stupid.

This is with wpilib v2025.1.1 and pathplanner v2025.0.0-beta-6

C++

The following code segfaults:

**Describe the bug**
```
    NamedCommands::registerCommand("my_command", frc2::cmd::Print("Hello"));
    std::cout << "Registered, now fetching" << std::endl;
    auto mycommandafter = NamedCommands::getCommand("my_command"); // Seg fault
```

Yields:
```
Registered, now fetching
Error: signal 11:
```

This stack may not be immediately useful:

(_Z7handleri+0x2e)[0x55571dfa91ae]
(+0x45320)[0x7fe22d655320]
(__libc_free+0x25)[0x7fe22d6bdd45]
(free+0x22)[0x55571e02fb32]
(_ZN3wpi19SmallPtrSetImplBaseD1Ev+0x33)[0x55571e0ce59b]
(_ZN3wpi15SmallPtrSetImplIPN4frc29SubsystemEED1Ev+0x1c)[0x55571e0ce7a2]
(_ZN3wpi11SmallPtrSetIPN4frc29SubsystemELj4EED1Ev+0x1c)[0x55571e0ce7c2]
(_ZN3wpi8SmallSetIPN4frc29SubsystemELj4ESt4lessIS3_EED1Ev+0x1c)[0x55571e0ce7e2]
(_ZN11pathplanner11CommandUtil19wrappedEventCommandESt10shared_ptrIN4frc27CommandEE+0x226)[0x55571e0cbf04]
(_ZN11pathplanner13NamedCommands10getCommandENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x91)[0x55571e0da24f]

Based on some light poking around it feels like the command is being deallocated before the wrappedEventCommand call is trying to poke around in its requirements... but I'm not the most versed on the command based implementation.

Any help would be appreciated!

michaelgtodd, 2025-01-07:
So I still don't know -exactly- what causes this, after alot of investigation the commands seem to be still allocated and they're being properly moved, but interactions with requirements objects (getrequirements, etc) cause a segfault.

However, when moving the source and header files of pathplanner lib directly into our project (rather than installing the static library and linking against it) the issue doesn't occur.

Purely speculation, it almost feels like the version of the wpi::smallset being built into the .a is different than that in our main application... but wpilib is installed and there's no other copy of it on the system path, so I don't see how that would be possible.

In any case, there doesn't appear to be any issue with pathplanner lib and we're now able to run paths and receive events as expected. Closing this issue.

closed 2025-01-07 by michaelgtodd

mjansen4857, 2025-01-07:
So the issue was probably because you were using WPILib 2025.1.1 and pplib beta 6. Pplib beta 6 was not built for that version hence the mismatch. If you use pplib 2025.1.1 that should also work.

michaelgtodd, 2025-01-07:
We were building the whole kit and kaboodle from source though (wpilib and PP), so I wouldn't imagine that should matter?

Though I haven't upgraded to the 2025 release yet, I can give that a shot tomorrow.

mjansen4857, 2025-01-07:
Unless you’ve copied the entire PPLib source into your project instead of using the vendordep then your robot program is just linking against a prebuilt PPLib library.

michaelgtodd, 2025-01-07:
This is all running native on Linux, in ROS nodes actually, ie there is no robot project, no wpilib vs code, no vendor deps, so it's all a quite atypical environment.

mjansen4857, 2025-01-07:
Ah ok. No idea then.