mjansen4857/pathplanner #971: v2025.1.1 dmg cannot be opened on mac arm64

jlmcmchl, 2025-01-05:
**Describe the bug**
Attempted to install the kickoff release of pathplanner, but I am unable to open the application because it could not be verified to be free of malware.

[image]

mjansen4857, 2025-01-05:
This is a mac security thing. You have to go the the security section in settings and click the "open anyway" button

AustinC02, 2025-01-08:
There is also a quick work around where you hold control and click on the application. In the menu that appears press open. After that, you should be able to open regardless of the security feature.