mjansen4857/pathplanner #993: path planner will not open on mac

team8845, 2025-01-10:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Versions: (please complete the following information):**
 - OS: [e.g. Windows 11]
 - GUI Version: [e.g. 2024.1.1]
 - PPLib Version: [e.g. 2024.1.1]
 - PPLib Language: [e.g. Java]

**Additional context**
Add any other context about the problem here.

team8845, 2025-01-10:
v2025

team8845, 2025-01-10:
> 
> When I open path planner this is what  I get is this uncaught error image 
> Please provide more information/fill out the issue template
w￼

mjansen4857, 2025-01-10:
There is no image attached