mjansen4857/pathplanner #996: Using FollowPathCommand directly causes segfault

michaelgtodd, 2025-01-11:
**Describe the bug**
The robot project at: https://github.com/michaelgtodd/pathplannersegfaultexample demonstrates an issue where when using FollowPathCommand directly a segfault can be encountered. 

This is an atypical integration of the pathplanner library, so it could certainly be something I'm doing. 

It's worth noting I've only seen this when a sequential command group is used in the path in question, but there could be other scenarios. 

The command inside of the sequential command group actually does execute and print, but the segfault -seems- to happen immediately after.

**To Reproduce**
Clone the robot project above
Enter simulation mode
Start autonomous mode
Wait a couple seconds and the following assertion should hit:

Assertion failed: idx < size(), file D:\a\allwpilib\allwpilib\wpiutil\src\main\native\thirdparty\llvm\include\wpi/SmallVector.h, line 307

**Expected behavior**
No segfault is encountered and the path continues as expected

**Versions: (please complete the following information):**
WPILib 2025.1.1
PathPlanner 2025.2.1

michaelgtodd, 2025-01-12:
Verified this still occurs on WPILib 2025.2.1 the full stack is something like:
[image]

calcmogul, 2025-01-15:
Here's the relevant code in SequentialCommandGroup:
```cpp
SequentialCommandGroup::SequentialCommandGroup(
    std::vector<std::unique_ptr<Command>>&& commands) {
  AddCommands(std::move(commands));
}

void SequentialCommandGroup::Initialize() {
  m_currentCommandIndex = 0;

  if (!m_commands.empty()) {
    m_commands[0]->Initialize();
  }
}

void SequentialCommandGroup::Execute() {
  if (m_commands.empty()) {
    return;
  }

  auto& currentCommand = m_commands[m_currentCommandIndex]; // line 32 from the backtrace
```

michaelgtodd, 2025-01-15:
Also tried a version that places the FollowPathCommand on the scheduler, seems to yield the same fault.

https://github.com/michaelgtodd/pathplannersegfaultexample/tree/schedule_path_command

michaelgtodd, 2025-01-15:
It's also worth noting that as laid out in the path json file, the sequential command structure only has a single named command in it, and this fault occurs after that command is executed (in the example all it does is print '1_command' but that print does occur before the segfault

michaelgtodd, 2025-01-15:
@calcmogul 's proposed changes seemed to resolve the issue in a local test setup.

PR #1012 (2025-01-16): Use references in EventScheduler for loops
Resolves #996
closed 2025-01-16 by mjansen4857