mjansen4857/pathplanner #997: AutoBuilder.configure breaks pyfrc unit tests

shou692199, 2025-01-11:
**Describe the bug**
With only one test enabled, it never ends.
https://pastebin.com/vkNFSzLT

With more than one test, I got `AutoBuilder has already been configured. This is likely in error.` and GC failure.
https://pastebin.com/GxzjcPVz

With AutoBuilder.isConfigured() condition statement, I still got GC failure.
https://pastebin.com/yymqWJ0w

Nothing wrong with `robotpy sim`

**To Reproduce**
Steps to reproduce the behavior:
1. Create a robotpy project with pathplannerlib and configure it in swerve subsystem
2. Run `robotpy test`

**Expected behavior**
Test pass.

**Screenshots**
[image]

**Versions: (please complete the following information):**
 - OS: Windows 11
 - GUI Version: 2025.2.1
 - PPLib Version: 2025.2.1
 - PPLib Language: Python

mjansen4857, 2025-01-11:
You should probably reach out to someone from robotpy who might be better able to help with this. I don't know too much about the robotpy/python testing setup so I can't be too much help, and also don't know how it would be a PPLib issue if it works fine normally.

shou692199, 2025-01-12:
From what I found, PathPlanner makes the Phoenix library initialization process take significantly more time than usual.

shou692199, 2025-01-12:
After moving the AutoBuilder configuration logic to the RobotContainer, everything seems to be working fine now.