mjansen4857/pathplanner #1001: Path Planner Incorrectly follows trajectory only on red alliance

alexanderszura, 2025-01-12:
**Describe the bug**
A clear and concise description of what the bug is.
The robot doesn't follow the trajectory when using the red alliance flipped path

**Screenshots**
[image]

**Versions: (please complete the following information):**
 - OS: [Windows 11
 - GUI Version 2025.2.1
 - PPLib Version: 2025.2.1
 - PPLib Language: Java

bobsfriend12, 2025-01-14:
I'm also having this issue. When following the flipped path (red alliance) it always veers to the left, just strafe no rotation. 

Heres my example with my path. Its just supposed to drive straight forward. In the upper left is my swerve module states. Red is the commanded swerve states and green is the commanded chassis speed straight from path planner. The bottom left shows that path planner is correctly setting the odometry before the auto, and the bottom right shows path planners telemetry screen. I show both the blue working and the red not working. I also tested with various other path shapes.

https://github.com/user-attachments/assets/2d803e20-6ffa-4966-af61-d1809b6bc923

Version:

OS (Driverstation): Windows 10
OS (Simulated): Macos Sequoia 15.1.1
GUI: 2025.2.1
PPLib: 2025.1.1
PPLib Lang: Java

Our code can be found [here](https://github.com/team2338/frc2025
)

mjansen4857, 2025-01-14:
@alexanderszura can you verify that you’re actually using PPLib 2025.2.1 by checking the version in the vendordep json. If so, can you provide some more info on what it’s actually doing/share code?

@bobsfriend12 update to PPLib 2025.2.1

bobsfriend12, 2025-01-14:
@mjansen4857 that fixed it for me thanks!

alexanderszura, 2025-01-14:
@mjansen4857 
[image]

alexanderszura, 2025-01-14:
Well this is awkward... It is working now. Maybe it was cached somewhere and when I ran a clean build it cleared the cache. Sorry about that

mjansen4857, 2025-01-14:
No problem, at least it works. That seems to happen sometimes for some reason.

closed 2025-01-14 by mjansen4857