mjansen4857/pathplanner #998: I'm unable to GIF my autos

980-Programming-Team, 2025-01-12:
**Describe the bug**
I tried to create a GIF of a auto and a error happened so I tried again, this time with running as administrator, the same thing happened. I tried creating a GIF of a path and it worked perfectly fine this time but then I tried creating an auto again and I got the same error. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to PathPlanner GUI / Application
2. Load up any auto and click on the export button at the top right
3. Select GIF and any color mode (I only tried in Dark so this could be a potential issue I'm unsure)
4. And then the encoding process should fail and you should get an error. 

**Expected behavior**
I was expecting that the encoder process not fail and I would be given the option to save my GIF to my desired location.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Versions: (please complete the following information):**
 - OS: Windows 11
 - GUI Version: 2025.2.1
 - PPLib Version: 2025.2.1
 - PPLib Language: Java

**Additional context**
Stack Trace Error:
#0      PngDecoder.decode (package:image/src/formats/png_decoder.dart:492)
#1      _TrajectoryRenderDialogState._encodeGif.<anonymous closure> (package:pathplanner/widgets/dialogs/trajectory_render_dialog.dart:311)
#2      IsolateManagerFunction.customFunction.<anonymous closure> (package:isolate_manager/src/isolate_manager_function.dart:109)
#3      _RootZone.runUnaryGuarded (dart:async/zone.dart:1609)
#4      _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:366)
#5      _DelayedData.perform (dart:async/stream_impl.dart:542)
#6      _PendingEvents.handleNext (dart:async/stream_impl.dart:647)
#7      _PendingEvents.schedule.<anonymous closure> (dart:async/stream_impl.dart:618)
#8      _microtaskLoop (dart:async/schedule_microtask.dart:40)
#9      _startMicrotaskLoop (dart:async/schedule_microtask.dart:49)
#10     _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:118)
#11     _RawReceivePort._handleMessage (dart:isolate-patch/isolate_patch.dart:185)

mjansen4857, 2025-01-12:
I've seen this issue before, and it doesn't happen on every auto. It can happen on some paths as well. There is some condition that makes it fail, but I have no idea what it actually is. Unfortunately, this issue happens within a library that PP uses to encode the GIF, so it's not something I'm capable of fixing at the moment. If I can figure out exactly what causes it, I should be able to open in issue in the encoding library. But, from everything I've seen so far it seems like it's almost random/system dependent.

Can you:
1. Provide the exact path files/auto file that cause this issue for you
2. Give me system specs of the machine you're running on, specifically RAM size and CPU

mjansen4857, 2025-01-12:
Do you have HDR enabled in your system/display settings?