mjansen4857/pathplanner #1002: PathPlanner Crashes when Attempting to Export a Choreo Path as GIF

Levlio123456, 2025-01-13:
**Describe the bug**
PathPlanner gives the error `RangeError (length) Invalid value: Not in inclusive range 0..49189: 49286` while attempting to export a choreo path as a GIF. Only happens sometimes.

**To Reproduce**
Steps to reproduce the behavior:
1. Create new auto.
2. Import choreo path.
3. Export auto as Gif.
4. See error

**Expected behavior**
It should export a GIF of the Auto.

**Screenshots**
[image]

**Versions**
 - OS: Windows 10
 - GUI Version: 2025.2.1
 - PPLib Version: 2025.2.1
 - PPLib Language: Java
 - Choreo Version: 2025.0.1

Levlio123456, 2025-01-13:
Also, the stack trace is:
`
#0      _Uint8ArrayView.[] (dart:typed_data-patch/typed_data_patch.dart)
#1      InputBuffer.readUint32 (package:image/src/util/input_buffer.dart:194)
#2      PngDecoder.decodeFrame (package:image/src/formats/png_decoder.dart:300)
#3      PngDecoder.decode (package:image/src/formats/png_decoder.dart:492)
#4      _TrajectoryRenderDialogState._encodeGif.<anonymous closure> (package:pathplanner/widgets/dialogs/trajectory_render_dialog.dart:311)
#5      IsolateManagerFunction.customFunction.<anonymous closure> (package:isolate_manager/src/isolate_manager_function.dart:109)
#6      _RootZone.runUnaryGuarded (dart:async/zone.dart:1609)
#7      _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:366)
#8      _DelayedData.perform (dart:async/stream_impl.dart:542)
#9      _PendingEvents.handleNext (dart:async/stream_impl.dart:647)
#10     _PendingEvents.schedule.<anonymous closure> (dart:async/stream_impl.dart:618)
#11     _microtaskLoop (dart:async/schedule_microtask.dart:40)
#12     _startMicrotaskLoop (dart:async/schedule_microtask.dart:49)
#13     _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:118)
#14     _RawReceivePort._handleMessage (dart:isolate-patch/isolate_patch.dart:185)

`

mjansen4857, 2025-01-14:
Sort of a duplicate of #998 

Different errors, but same situation

closed 2025-01-14 by mjansen4857