mjansen4857/pathplanner #1009: It crashes when opening project on macOS

Hirose-Yuto, 2025-01-15:
**Describe the bug**
The application crashes when opening a robot project.

**To Reproduce**
1. Open PathPlanner app.
2. Click "Open Robot Project"
3. Select a Robot Project
  i. This happens for any project
4. It crashes immediately

When recording a bug, I noticed that the error appears only for a moment since it crashed immediately.
```
PlatformException （UnexpectedError, Error while creating bookmark Error Domain=NsCocoaErrorDomain Code=256 "Failed to retrieve app-scope key"
Userinfo=/NSDebugDescription= Failed to retrieve app-scope Key｝ for file:///Users/hirose/Dropbox/Mac/~~~
```

**Expected behavior**
The project is opened.

**Screenshots**
[image]

**Versions: (please complete the following information):**
 - OS: macOS Sequoia (15.0.1)
 - Tip: Apple M2 Pro
 - GUI Version: 2025.1.1 and 2025.2.1
 - PPLib Version: 2025.2.1
 - PPLib Language: Java

**Additional context**
I worked well 2 days ago, but this happened today.
As far as I remember, I did nothing but edit, build and simulate codes.

My English may be a little strange because I am Japanese.
Thank you!

mjansen4857, 2025-01-15:
There's probably some problem with it being in your Dropbox folder.

Here's a few things to try:
1. Run `defaults delete com.mjansen4857.pathplanner` in the terminal to reset app settings
2. Check in System Settings -> Privacy & Security -> Files & Folders and make sure that you don't have any folder access turned off for PathPlanner
3. Open a project outside of a dropbox folder, iCloud folder, or any other cloud storage folder like that

Hirose-Yuto, 2025-01-16:
As a conclusion, it works now.

What I did
1. Run `defaults delete com.mjansen4857.pathplanner` 
2. Open PathPlanner & open folder
3. Successfully opened

I run `defaults delete com.mjansen4857.pathplanner` based on other issues yesterday, so I don't think it helped.
I have no idea, but I would try to move a project outside of a dropbox folder if the same issue happened.

Thank you for your coorporation.

btw, PathPlanner cannot be found in System Settings -> Privacy & Security -> Files & Folders or Full Disk Access.

closed 2025-01-16 by Hirose-Yuto