mjansen4857/pathplanner #1013: Cannot edit path

5690Programmers, 2025-01-16:
**Describe the bug**
I am unable to edit any path or add a path to my auto. When I attempt to do so, the GUI crashes.

**To Reproduce**
Steps to reproduce the behavior:
1. Open pathplanner
2. Create path
3. Click on path or preexisting path
4. Encounter Uncaught Error "Unsupported operation: Infinity or NaN toInt"

**Expected behavior**
The pathplanner path editor 

**Versions: (please complete the following information):**
 - OS: Windows 11 Pro version 24H2 build 26100.2605
 - GUI Version: 2025.2.1
 - PPLib Version: 2025.2.1
 - PPLib Language: C++

**Additional context**
Stack Trace:
```
#0      double.~/ (dart:core-patch/double.dart)
#1      _RepeatingSimulation.x (package:flutter/src/animation/animation_controller.dart:1009)
#2      AnimationController._startSimulation (package:flutter/src/animation/animation_controller.dart:832)
#3      AnimationController.repeat (package:flutter/src/animation/animation_controller.dart:746)
#4      _SplitPathEditorState._simulatePath (package:pathplanner/widgets/editor/split_path_editor.dart:772)
#5      _SplitPathEditorState.initState.<anonymous closure> (package:pathplanner/widgets/editor/split_path_editor.dart:127)
#6      SchedulerBinding._invokeFrameCallback (package:flutter/src/scheduler/binding.dart:1397)
#7      SchedulerBinding.handleDrawFrame (package:flutter/src/scheduler/binding.dart:1331)
#8      SchedulerBinding._handleDrawFrame (package:flutter/src/scheduler/binding.dart:1176)
#9      _rootRun (dart:async/zone.dart:1414)
#10     _CustomZone.run (dart:async/zone.dart:1317)
#11     _CustomZone.runGuarded (dart:async/zone.dart:1225)
#12     _invoke (dart:ui/hooks.dart:314)
#13     PlatformDispatcher._drawFrame (dart:ui/platform_dispatcher.dart:427)
#14     _drawFrame (dart:ui/hooks.dart:283)
```

5690Programmers, 2025-01-17:
We just found that editing the gearing on the robot causes the issue. Initially it was 5.143 and I was able to edit the path. However, when I changed it to 4.5 it began causing problems. We have isolated the issue to being only causes by changing that one value.

mjansen4857, 2025-01-17:
This is probably because you a creating a robot configuration that is not capable of actually accelerating the robot. If any of the calculated values under the robot preview in the settings are red then your config is probably bad.

mjansen4857, 2025-01-17:
I can prevent this crash from happening, but the underlying issue of the path not generating will remain which would be due to the config or control point placements.

5690Programmers, 2025-01-17:
I feel there should be some more useful way to indicate that the config is invalid rather than just throwing you a stack trace? To note however, we started changing values around and then when eventually we changed it to the value that initially didn't work, it worked. We checked the git diff for our project and everything and it just appeared to magically work. This could be some super strange bug with something other than pathplanner but I doubt the underlying problem will be found since this seems extremely difficult to reproduce. Thank you for the quick response!

closed 2025-01-17 by mjansen4857