mjansen4857/pathplanner #1019: Memory Leak on ADStar Planning

scarmain, 2025-01-18:
**Describe the bug**
I was looking at memory usage on the Rio 1 for our robot, and it looks like ADStar planning is going crazy allocating objects.  I'm writing this down just to document it, not sure how to diagnose further.  It seems to be allocating 4x more objects per second from our robot code

[image]

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. I followed the steps to connect to the robot with VisualVM (https://docs.wpilib.org/en/stable/docs/software/advanced-gradlerio/profiling-with-visualvm.html#analyzing-a-heap-dump)  This could connect to simulation too, but my screenshots are from the robot.
2. Code at https://github.com/FRC2832/Robot2832-2025Kitbot, but the snapshots were taken while disabled, no driving started.  No team code directly called ADStar to my knowledge (we do have calls to PathPlanner DriveToPose, but not executed yet when this was measured).

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Versions: (please complete the following information):**
 - OS: [e.g. Windows 11]
 - GUI Version: [e.g. 2024.1.1]
 - PPLib Version: [e.g. 2024.1.1]
 - PPLib Language: [e.g. Java]

**Additional context**
Add any other context about the problem here.

Baconing, 2025-02-28:
I also have this, did you ever fix it or are you still waiting for somehting?

mega12345mega, 2026-03-22:
From what I can tell this is caused by [this line in LocalADStar#runThread](https://github.com/mjansen4857/pathplanner/blob/e02bbf3176588166e8fe5192ab8dea85f6d62f7a/pathplannerlib/src/main/java/com/pathplanner/lib/pathfinding/LocalADStar.java#L273), which copies the entire `requestObstacles` `HashSet` every 10ms regardless of if it is actually required.

Since the navgrid.json file I tested with had 530 obstacles, this results in 53,000 HashMap entries per second.

Could be fixed by moving the copy to before [this line](https://github.com/mjansen4857/pathplanner/blob/e02bbf3176588166e8fe5192ab8dea85f6d62f7a/pathplannerlib/src/main/java/com/pathplanner/lib/pathfinding/LocalADStar.java#L288) (also requires moving the unlock) or simply replacing `requestObstacles` every time it is written (do `requestObstacles = new HashSet<>()` instead of `requestObstacles.clear()`; it is always cleared before being written to)