mjansen4857/pathplanner #1021: Access is denied

Itay2200Ba, 2025-01-19:
**Describe the bug**
I downloaded Path planner through the microsoft store and since i frist tried to enter the app it showed a message saying access denied
**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**

[image]

**Versions: (please complete the following information):**
 - OS: [e.g. Windows 11]
 - GUI Version: [e.g. 2024.1.1]
 - PPLib Version: [e.g. 2024.1.1]
 - PPLib Language: [e.g. Java]

**Additional context**
Add any other context about the problem here.

mjansen4857, 2025-01-20:
You selected a project location that the OS prevents access to. It should let you pick another if it fails to open but I must have missed somewhere. Please provide the stack trace.

totNate, 2025-01-23:
I am also experiencing this error and I as well can not open the app to fix it

totNate, 2025-01-24:
this seems to me that it is the same as #1020

closed 2025-01-27 by mjansen4857