mjansen4857/pathplanner #1027: Field Flipping Mirroring

alexanderszura, 2025-01-23:
I don't like wasting time creating two autos that will do the same thing but are mirrored. For example, during this season (2025), creating an auto path for coral that can be mirrored so that you don't have to create multiple auto paths and autos would be awesome.

Either,
A) A boolean supplier similar to field mirroring for red/blue alliance.
B) A button that will automatically mirror a path/auto in path planner
C) An option to do it "Manually" with the field mirroring class.

Something like this I made for an auto align project
`public static Pose2d mirrorPose(Pose2d pose) {
        return new Pose2d(new Translation2d(pose.getX(), FlippingUtil.fieldSizeY - pose.getY()), new Rotation2d(Math.PI * 2).minus(pose.getRotation()));
}`

blaze-developer, 2025-01-25:
I second this.

mjansen4857, 2025-01-26:
Already exists for [paths](https://pathplanner.dev/api/java/com/pathplanner/lib/path/PathPlannerPath.html#mirrorPath()) and [autos](https://pathplanner.dev/api/java/com/pathplanner/lib/commands/PathPlannerAuto.html#%3Cinit%3E(java.lang.String,boolean))

closed 2025-01-26 by mjansen4857

blaze-developer, 2025-01-26:
Is this already in the release version? If it is, I cant find that constructor in my project. I am using 2025.2.1.