mjansen4857/pathplanner #1030: 2025 Reefscape field image / Processor alignment reference off by 36 pixels for both alliances

roberttuttle, 2025-01-26:
The visual alignment marker present in the field image for both processors is off by 36 pixels for both alliances as compared to the pose coordinates to the center of the AprilTag for each. It appears that those markers were drawn / placed to have their center on the start of the processor station as opposed to representing the center of the processor scoring element (which is represented by the associated AprilTag center).

roberttuttle, 2025-01-27:
It looks like the Blue processor marker in the image is also off, so I will try to help out by fixing both and submitting a PR.

mjansen4857, 2025-01-27:
I’ll be able to get to fixing the images fairly soon. To accept a PR for that I’d want the source assets to be updated as well, which uses affinity photo. If you don’t have that then I can fix them, it’s a pretty minor change.

roberttuttle, 2025-01-27:
Cool ... no problem waiting for the source of truth fixes then. I wanted to offer to fix it since "complaining" about 36 pixels in the grand scheme of things is obviously minor. 😉

closed 2025-01-27 by mjansen4857