mjansen4857/pathplanner #1031: Path Planner Tabs, and Back and Forward Gestures

michaelk036, 2025-01-26:
I have been having issues with _PathPlanner_, specifically not being able to switch between multiple Paths and Autos, while being able to go back and forth in 1 mouse click. This becomes really frustrating over time. Also, my mouse has back and forward buttons, and whenever I click them they don't work within _PathPlanner_, which is also pretty frustrating.

The solution I would like for this is Web Browser - like tabs, which allow switching between multiple `.path` or `.auto` files. VS Code or IntelliJ or basically any other modern IDE has this as well. At least to me, this seems like a pretty simple feature that got over looked by the developers of this. Also, similar to a web browser back and forward buttons, and mouse back and forward gesture implementation would be fantastic. 

Here's what I would ideally want:
[image]

Thank you, and I hope this gets implemented at some point. 
-Michael, Team 4048

JAMthepersonj, 2025-01-26:
I have run into that same thought many a times when using PathPlanner and I think it would be very beneficially to the overall experience of using PathPlanner and make PathPlanner more of a streamlined process

mjansen4857, 2025-01-27:
I like the idea of tabs, but it's definitely not a simple feature to implement. It would require a pretty major restructuring to get it to work. So, this will be a feature I'll target for next year.

michaelk036, 2025-01-28:
Thankyou, I hope this feature gets implemented at some point before next season. 😁