mjansen4857/pathplanner #1035: Compatibility to older fields (FieldSymmetry)

aaronleetw, 2025-01-27:
**Is your feature request related to a problem? Please describe.**
It would be convenient for new PathPlannerLib versions to be compatible with older games. The main issue preventing this is the "non-standard" field mirroring adopted by CRESCENDO (2024) and Charged Up (2023). Since the new Reefscape (2025) season, PPLib has reverted to rotational mirroring to be compatible with the field. Unless I am mistaken, there is no public function available that allows users to configure this behavior.

**Describe the solution you'd like**
Add an AutoBuilder constructor that allows users to configure the `FieldSymmetry` type used by `FlippingUtil`. The default behavior would still use the current mirroring, so there should be no disruption to current teams.

**Additional context**
c96faa2301393c4ce04b53ec846a854d2c9b0a98 introduced [`FlippingUntil`](https://github.com/mjansen4857/pathplanner/blob/main/pathplannerlib/src/main/java/com/pathplanner/lib/util/FlippingUtil.java) using `FieldSymmetry.kMirrored`.
e749a1d9b760dc45f6755f538bd36eed15d9ea10 switched to `FieldSymmetry.kRotational`.

I can work on this if this feature request is accepted. Thank you for your hard work!

mjansen4857, 2025-01-27:
The type of mirroring and the dimensions of the field are public fields in FlippingUtil, which allows the user to customize them to work for different fields. So, just set them to the values for whatever field you want before anything actually gets flipped.

aaronleetw, 2025-01-27:
I was mistaken. Thank you for your reply!

closed 2025-01-27 by aaronleetw