mjansen4857/pathplanner #1050: Question: What is the Local AD* pathfinding algorithm? Are there any resources about it?

ancientstraits, 2025-02-03:
I know about what the A\* Pathfinding algorithm is, since that is a well-known pathfinding algorithm. However, the algorithm that Pathplanner uses, "Local AD\*" , has no documentation on the Internet besides the PathPlanner docs.

The beginning of `LocalADStar.java` says "I would like to apologize to anyone trying to understand this code. The **implementation I translated it from** was much worse." This means that there is another implementation of Local AD*, outside of PathPlanner.

Where can I find more information about this algorithm? Or is it actually completely custom-made for PathPlanner?

mjansen4857, 2025-02-03:
The "Local" part just means it's running on the roboRIO. Its described on page 10 of [this paper](https://www.cs.csustan.edu/~mmartin/teaching/CS4960S15/Corey%20Trevena%20-%20Pathfinding%20Algorithms%20in%20Navigational%20Meshes%20PDF.pdf)

ancientstraits, 2025-02-03:
Thank you so much for your answer?

> The "Local" part just means it's running on the roboRIO

Has anyone worked on a `Pathfinder` that runs on a coprocessor instead? I am asking because I am currently trying to make a program that offloads `LocalADStar` pathfinding to a coprocessor.

mjansen4857, 2025-02-03:
There was one available last year, but I've abandoned it because it doesn't actually provide enough of a performance improvement to bother maintaining it. It may be worth it on a RIO 1, but for a RIO 2 it's fast enough to just run locally.

ancientstraits, 2025-02-03:
That's great! Where can I find it?

ancientstraits, 2025-02-03:
@mjansen4857 I found the `LocalADStar.java` file in a 2024 commit, but I do not know where the code that should be placed on the coprocessor is located.

ancientstraits, 2025-02-04:
I found the code for the coprocessor at https://github.com/mjansen4857/pathplanner. Thank you so much!