mjansen4857/pathplanner #1053: PathfindingCommand Completing When Robot is Within 0.5 Meters of Target Pose

Murat65536, 2025-02-07:
Upon the initialization of the PathfindingCommand, the robot checks if it is within 0.5 meters of the target pose. If it is, it sets the variable finish to true. This skips execute and returns true in isFinished. This is an issue since our team was planning to use PathPlanner's PathfindingCommand to visit different poses on the reef, and we simply cannot be within 0.5 meters of our intended pose and still be able to score.

Steps to reproduce the behavior:
1. Run the PathfindingCommand while close to your intended destination.

We expected this to be able to reach our pose. Instead, the robot thought that it was close enough and didn't move.

[image]

 - OS: Windows 11
 - GUI Version: 2025.2.2
 - PPLib Version: 2025.2.3
 - PPLib Language: Java

I believe that creating a variable that can be changed in the Pathfinding class and then calling it in PathfindingCommand instead of having a static number could resolve this issue. It would preserve the default value while still allowing modification.

mjansen4857, 2025-02-07:
This is intended. Changing that value doesn’t solve the underlying issue that it is there to prevent in the first place. The pathfinder cannot generate very short paths like that because the start and end position can be the same node on the navigation grid. Basically it already starts at the goal. 

Don’t use pathfinding for short distance paths or if you know there won’t be an obstacle in the way, use on the fly paths for that instead, which is much more efficient and controllable.

Murat65536, 2025-02-07:
Since PathPlanner allows for changing the number of nodes covering a grid, could this at least be altered to scale with that amount? This would ensure that the start and end positions are not on the same node while bringing the robot closer to its desired destination.

prensing, 2025-04-06:
This should be documented. Having a hard coded cut in code which is **not** clearly indicated in the documentation is definitely not "best practices".