mjansen4857/pathplanner #1094: Event markers won't trigger

Tricks1228, 2025-03-22:
When the path is running, event triggers at the end sometimes wont trigger, causing the rest of the auto to stall.  Triggers that fail are foud at the end positions of the path. Testing with our code shows that it has failed most commonly when the position of the event marker is set to 1 but it has failed as well with the event trigger at 0.99.

To Reproduce:
1. Create an event trigger on a path and set the position to be at the end, or very close to the end of a path
2. Run the path, the bugged behavior appears to happen at random however.

Expected Behavior:
The path should be continuing cleanely onto its next scheduled path, however when the event trigger doesn't get triggered then the entire auto stalls.

Code for Event Markers:
`        new EventTrigger("intake coral").onTrue(Routines.intakeCoral(elevator, intake));
        new EventTrigger("score L4").onTrue(Commands.waitUntil(() -> elevator.nearHeight(Positions.L4.close)).andThen(Routines.scoreCoralLevelFour(elevator, intake)).withName("Auto score L4")); // ERROR HERE
        new EventTrigger("raise elevator to intake").onTrue(elevator.setPositionCommand(Positions.INTAKE.close));
        new EventTrigger("raise elevator to L4").onTrue(elevator.setPositionCommand(Positions.L4).alongWith(intake.coralIntake()));
        new EventTrigger("stow elevator").onTrue(elevator.setPositionCommand(Positions.STOW));
        new EventTrigger("test").onTrue(Commands.print("test event trigger"));`

Versions:
 - OS: Windows 11
 - PPLib Version: 2025.2.6
 - PPLib Language: Java

Additional Context:
We have verified that .nearHeight() is returning true when we want it to.

w-reynolds, 2025-03-24:
Tip:
When including blocks of code in a Github Issue, start with a line of just three of the grave accent or back tick symbols. (The key to the left of the 1 Key).

Then paste your code.

At the end of your block of code include another line with three of the grave accent or back tick symbols to revert back to general text.

This way the carriage returns will be honored and it will make reading your code a lot easier for anyone trying to help you.  
I'm guessing you have only used one of the symbol at the start and end of your code instead of three, that is why your code is wrapping.

Single grave accent symbols are generally used for talking about a variable or small section of code inline in a sentence
like `m_myVariable` was never initialized.

Also clicking on the preview tab will show what the post will look like before you commit the post, you can go back to editing it by clicking on the Write tab.

w-reynolds, 2025-03-24:
Hi, 
I'm not much of programmer, and we use a different language C++, and I'm usually looking for help here rather than offering any suggestions but I have some thoughts on your problem.

Firstly in your Github code I noticed that you have a "drive to coral" named command, this is used by your 2 coral, I K auto, If the Swerve.drive Command requires your drive subsystem, I thing this will interrupt your auto because it also requires the drive subsystem, which will cause your auto to stop.  I could be completely wrong, could this be related to your problem?

Also, I'm not sure how the new event triggers are processed in the path planner source code, I have a vague recollection from looking at the path planner source years ago (when it was a lot less complicated) that two parallel command groups were running for each path segment being followed. One for the path following (requiring the DriveSubsystem) and another one for the named command processing, with a timer being polled to fire off the named commands as the robot traveled along the path.

I think it use to work like this, if this timer is greater that the trigger point value times the taken to finish the path, it would schedule the command.  You say your problem is intermittent, maybe the last execution cycle of the path follower was at a point 0.989 along the path.  The robot is moving towards the goal and on the next execution cycle it is close enough to the end of the path to symbolize that it has finished the path.   Because the path following is finished there is no need to check for any more events on the path.

There is a change by putting your event very close to the end it might get missed because the path ends before your trigger go off.

I'm more than likely wrong, but it might explain why this works sometimes and not others.

If this is your problem, you could use named commands rather then triggered commands and then place the named command in the sequential command group in the pathplanner GUI, so that when the path is finished, it will definitely go onto the next command in the group.

Hope this helps, and I could be completely wrong, so don't take my ideas as gospel.  I thing the brains trust that usually monitor these issues are flat out with competitions and getting ready for champs, so I think we might be waiting a bit longer (understandable) to get an official response.  I hope you get it working for your event.

WispySparks, 2025-03-24:
Your intuition matches what I also think it might be, based on the intermittentness of the problem I believe it's timing related. I've looked into the source code and it seems that there could be a case where the path ends up finishing before triggering final event markers, as the `execute` is called, then some statements are run in the command scheduler, and `isFinished` is called, grabbing the time from a `Timer` again. For now we're going to use named commands and proxy them so we can still mix and match event triggers and named commands (we haven't had problems when the event triggers are early on like the one we use to raise our elevator)